bambulab/BambuStudio #2170: Exclude Objects feature - for the P1 series printers. No reason to have this price-walled.

nirurin, 2023-08-04:
Is your feature request related to a problem? Please describe.
Having to cancel a really long print that has multiple objects, only because one object failed. This is causing a huge loss of filament, time, electricity, money and frustration.
It's available for the X1 series, but I dislike that it's being price-walled. 

Describe the solution you'd like
Like klipper and octoprint, allow to cancel a failed object on the build plate, instead of cancelling the entire print. It obviously won't work from the printer controls, but it can be done via bambu handy or bambu studio. 

Describe alternatives you've considered
Using printers that support Klipper or Octoprint "Cancel Object" feature.

Additional context
https://www.klipper3d.org/Exclude_Object.html

npoulos, 2023-08-05:
Agree this feature needs to be available on the P1 series. It’s a software feature and there are clearly no hardware restraints. Especially since it’s been released after the fact and was never advertised as a coming soon feature for the x1. Feels like you’ve mislead p1 series buyers. 

Please enable this feature for the p1 series asap.

Hologos, 2023-08-07:
It depends on how they've implemented it. It may be computationally intensive and X1 series as way more powerful CPU. That's the reason for 0.5 FPS camera on P1 instead of 1080p fps camera on X1. If it uses some kind of tags / comments in the gcode to identify the regions to skip the objects, there should not be any reason not to port it for P1 series as well.

QingZhangBambu, 2023-08-08:
It is not pay-walled. P1serise has that function, but the screen does not support that. so we are figuring out a method to fix that. We will figure out that as fast as possible.

nirurin, 2023-08-08:
> It is not pay-walled. P1serise has that function, but the screen does not support that. so we are figuring out a method to fix that. We will figure out that as fast as possible.

I thought the X1 series already has a solution for that - you can use the bambu handy app to select objects to exclude. Why would this not work on the P1S?

antoan-boss, 2023-08-24:
@QingZhangBambu Hopefully it will be soon in Bambu Handy at least for p1 series :)

QingZhangBambu, 2023-08-24:
@antoan-boss @nirurin I will talk to the concerned staff about that.

nirurin, 2023-08-28:
> It is not pay-walled. P1serise has that function, but the screen does not support that. so we are figuring out a method to fix that. We will figure out that as fast as possible.

I'm being told (mostly by people who paid for the X1C) that the P1-series doesn't have the processing power to allow the exclude objects feature. Can you at least confirm that it's capable of doing it, and you're just working on the user interface / screen controls?

QingZhangBambu, 2023-08-29:
@nirurin  I can confirm that. This function will be implemented on P Series later. It has been put in our plan list. I do not responsible for UI, I work on Software main function implementation.

jostwald, 2023-09-27:
Hi, Is there any update on this? 

Thanks!

MDsza, 2023-10-01:
Yeah, I just fiddled in my G-Codes-File manually for 1 hour to get the print without 2 warped parts, just to see the nozzle crashing over the partly printed parts after resume ;-) . Just to exclude / skip such parts would be a 'live safe' for all the P1P/P1S 3D enthusiasts, too. So, c'mon bambu heroes: go for it :-)

Sn0w3y, 2023-10-25:
@QingZhangBambu would be great if you take care of this "Problem" as many users would be glad to have this feature and buy a P1S if it has this option !

MonkaKokosowa, 2023-10-27:
It seems that the newest firmware has exclude objects feature! It seems to work now as I just tested it on my P1P and it shows it on handy app.
[image]

Sn0w3y, 2023-10-27:
Well it says "does not work on MicroSD or LAN Printing) ? If i start the Print Job from BambuStudio will it work then?

MonkaKokosowa, 2023-10-27:
You can start a print job from either handy or slicer the important thing is that it goes through the cloud (dunno why tho). Though, at least for now, excluding objects can only be done in handy app (I suppose that will change with the next slicer update).

Sn0w3y, 2023-10-28:
Thats great News!

Can you start a job via Smartphone now aswell? Like Uploading a file from the Smartphone?

MonkaKokosowa, 2023-10-28:
No, when I said starting from the phone, I meant the printing history. There's probably some bug tracker issue for that, though I don't, to be honest, really see a use for it anymore after they added makerworld

SaltWei, 2024-01-18:
Excluding object can be used on P1 series now. Close

closed 2024-01-18 by SaltWei

lin-ycv, 2024-11-18:
> Excluding object can be used on P1 series now. Close

Except when in LAN mode... Can we please this feature implemented in BambuStudio, if it can be done thru the handy app, no reason why it can't be done in Studio

SaltWei, 2024-11-18:
> Except when in LAN mode... Can we please this feature implemented in BambuStudio, if it can be done thru the handy app, no reason why it can't be done in Studio

Because the screen of P1 is too small to implement the function. App is as "screen" of P series printer

lin-ycv, 2024-11-18:
Bambu studio reports basically all the same info as the handy app? 
Using bambu studio doesn't require interaction with the printer screen.

TheDuffman85, 2024-11-18:
> > Except when in LAN mode... Can we please this feature implemented in BambuStudio, if it can be done thru the handy app, no reason why it can't be done in Studio
> 
> Because the screen of P1 is too small to implement the function. App is as "screen" of P series printer

How exactly is the screen related to Bambu Studio? I call this BS.