bambulab/BambuStudio #2172: [Feature Request] "Ignore Colour" for Auto Refill

BambuLabDiscord, 2023-08-04:
Add a box to check so you can chose to "Ignore Colour" when using the same filaments but of different colour in Auto Refill. This would allow you to use up the tiniest bits of filament of different colours (but of the same type) when doing drafts or prints where the colour does not matter.

Merlwynd, 2023-08-04:
You magy know this already but you can accomplish this by specifying all spools are the same colour in the AMS section in studio or on the printer.  If you are using Bambu spools, you can do the same but just remove the RFID tag first…you are at the end of a spool anyway and you get a new RFID tag with a refill spool….

Misiu, 2024-02-17:
I have some leftovers of Bambu PETG. The only option for now is to manually remove the RFID tags from spools and configure the filament.
You should require the same material type or brand, but the color is irrelevant, there should be at least an option to allow different colors.

SnickCH, 2024-04-26:
+1 
Let manually chose the order of the slots (despite color or brand). 
Maybe two options to activate
- "ignore brand"
- "ignore color"

This is at the moment the most annoying part of the AMS with Bambu filament