bambulab/BambuStudio #2173: Vertical Layer Slider Right Click Menu Does Not Appear

candera, 2023-08-04:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

OS X Ventura 13.4.1 (c)

### Additional system information

2.4 GHz 8-Core Intel Core i9
AMD Radeon Pro 5500M 8 GB
Intel UHD Graphics 630 1536 MB
32 GB 2667 MHz DDR4

### Printer

X1-C

### How to reproduce

Prepare model as usual.
Preview
Attempt to "right click" (two-finger click on a trackpad, actually, since no right mouse button is available)
Observe that the slider scrolls down; no context menu appears.

### Actual results

The slider scrolls down; no context menu appears.

### Expected results

Context menu including "pause at this layer" should appear.

### Project file & Debug log uploads

[project.zip](https://github.com/bambulab/BambuStudio/files/12264014/project.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

lanewei120, 2023-08-05:
@candera 
thanks for reporting this issue,
could you give a snapshot of your issue?

candera, 2023-08-05:
Screen recording [here](https://photos.app.goo.gl/LuceKA5xm5aEWrBw6)

Since a screenshot of something NOT happening on a mouse click seemed less than useful. :)

xp-Yang, 2023-08-17:
You should right click on the round button so that the context menu will apear.

candera, 2023-08-17:
> You should right click on the round button so that the context menu will appear.

Aha! That works, thanks. However, it still moves the button. Like, I right-click (really, two-finger click) on it, and both the the button moves down and I get the context menu.

closed 2024-01-16 by LiZ-Li-BBL

Flamenco, 2024-07-02:
Hold down SHIFT when using 2-finger click and the button will not move.