bambulab/BambuStudio #2174: Update to version 1.7.2.51 brings error

copperfield74, 2023-08-04:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S

### How to reproduce

Trying to install and run into this error.
Installation is started as admin.
[image]

### Actual results

Installation doesn't succeed.

### Expected results

successful installation

### Project file & Debug log uploads

No log files

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rdekany, 2023-08-04:
I had the same issue, NVIDIA Broadcast was occupying the BambuSource.dll in the user\appdata folder. 

Solution:
1. Kill Nvidia Broadcast
2. Reinstall
3. Done

In case you have another process occupying the DLL, use Process Explorer to find out what it is:
Find >Find Handle or DLL > BambuSource.dll

copperfield74, 2023-08-05:
Thanks very much for your tip!
Process Explorer did the trick for me.
I don't know why but Skype was using the bambu process!?
I'm no developer, but this is strange and doesn't give me a good feeling. :-(

closed 2023-08-05 by copperfield74