bambulab/BambuStudio #2176: First layer defects detection does not allow resume

barkin-aygun, 2023-08-05:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 13.4.1

### Additional system information

MBP 14" 2021

### Printer

Bambu Carbon X1

### How to reproduce

Print something with first layer defect. First layer defect detection kicks in. Click close, and then try to click resume.

### Actual results

There is no resume button, pause and stop are the only buttons both on Device page of Bambulab AND on the device UI itself.

### Expected results

I should be able to see a resume button and press resume.

### Project file & Debug log uploads

[clips.3mf.zip](https://github.com/bambulab/BambuStudio/files/12265414/clips.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

dragon-eyes-bbl, 2023-08-05:
Hi @barkin-aygun , 
It could be a known issue with FM 1.6.0.0.
A new firmware was released hours ago, please reboot your printer and update to the latest firmware. 
Thanks

barkin-aygun, 2023-08-05:
@dragon-eyes-bbl i have updated earlier today to 01.06 so that can’t be it.

barkin-aygun, 2023-08-05:
Thanks i will try that.

closed 2023-08-05 by barkin-aygun