bambulab/BambuStudio #2177: Slicer printing in thin air

dtdionne, 2023-08-05:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 7

### Printer

X1c

### How to reproduce

Slice plate

### Actual results

It looks like the slicer is trying to print in thin air when it should be bridging.  Layer 360.

### Expected results

It should bridge.

### Project file & Debug log uploads

I can’t drop the zip here, it won’t paste…

### Checklist of files to include

- [ ] Log file
- [ ] Project file

dtdionne, 2023-08-05:
Here it is, layer 360, thanks

[ThinAir.zip](https://github.com/bambulab/BambuStudio/files/12266069/ThinAir.zip)

SaltWei, 2023-08-05:
[image]
This model has floating part. Same result on prusa.

SaltWei, 2023-08-05:
If you want such model can be printable. You can try two methods.
1 use manual paint support to support this layer. But maybe this method is not good enough. I havn't tried to printing, so not sure.

2 redesign the model to make it printable without support and floating part. You can only close the hole at the height off float part with only one layer thickness. After printing, you can easily remove the thin part. This is very common design method to make model printable.

dtdionne, 2023-08-06:
I disagree, it's a bottle cap with a hole.  There's no floating part anywhere and just because other slicers choose to try and print in thin air doesn't mean they're right either.  It could easily bridge diagonally instead of choosing to try and print in thin air.  What these slicers are doing is cumulative, a well known negative side effect of open source.  It's cascade failure, the first one does it and it got copied down the line to all the rest.  ANYTHING would be better than a printer EVER TRYING TO PRINT PLASTIC IN THIN AIR.  An error would be better, think of all the filament people could be wasting because of this!!!!  All that plastic and electricity wasted, and all that carbon ;)  Ur lucky im not some global climate change nut job. 

To be honest, it doesn't sound like many people have the chops to fix this in the actual code, another negative side effect of open source.

Bridging short segments would be light years better than bridging across the entire cap.  If i wasn't so heavily invested in this I'd try and solve this issue in the code because I bet there's a ton more BUGS where this BUG is...and it is a BUG.  Never should a slicer choose to try and print in thin air.

Bridging short segments would be way better than bridging across the entire cap.

https://youtu.be/GipMHJ9h5sM

SaltWei, 2023-08-08:
[image]

closed 2023-10-07 by SaltWei