bambulab/BambuStudio #2179: More 0.6 nozzle default profiles

capsel22, 2023-08-05:
**Is your feature request related to a problem? Please describe.**
I recently upgraded my standard 0.4 Nozzle to hardened steel 0.6 on my P1P and discovered BambuStudio only has one default profile on 0.30 layer height. The 0.4 nozzle has 7 predefined profiles.

**Describe the solution you'd like**
I'd like BambuLabs to provide more default profiles for 0.6 nozzle in line with 0.4

**Describe alternatives you've considered**
The only real alterative is to adjust layer height manually. However I do not have the knowledge or experience to adjust other parameters which go hand in hand with layer changes

capsel22, 2023-08-05:
Just noticed the P1S has more 0.6 profiles but not P1P? Could they just be moved to P1P?
P1S 0.6
[image]

P1P 0.6
[image]

QingZhangBambu, 2023-08-08:
We are testing the profile of P1P. It would be added soon.

ALEX-9-99, 2023-08-31:
The P1S has more profiles than the P1P, but they are only copied from the X1C and always give problems.

The last fix in the Gcode where the P1P drove the print bed into the bottom of the printer without a bottom stop is fixed on the P1P profiles, but still occurs on the X1C profiles.

closed 2025-05-30 by QingZhangBambu