bambulab/BambuStudio #2187: Send to Printer stating "not same LAN" when printer is in same lan

NeoTech, 2023-08-05:
### Bambu Studio Version

1.7.1.62

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab X1 Series

### How to reproduce

Change dropdown on preview page to "Send" and click the send button the popup window give you a error about the printer not being on the same LAN. Even if its clearly in the same ip range as my desktop computer. (192.168.x.x range)

### Actual results

[image]

### Expected results

Being able to send files to the device for storage on either SD card or internal storage..

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12268243/log.zip)
[Basegrid.zip](https://github.com/bambulab/BambuStudio/files/12268244/Basegrid.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2023-08-18:
Hi.@NeoTech 
This may be due to recording the old IP address.
Please close studio and delete these files before trying again
[image]

ianderl, 2023-08-27:
I have the same issue. Where is this folder location you posted a screenshot of?

walterwongbbl, 2023-08-29:
> I have the same issue. Where is this folder location you posted a screenshot of?

[image]

walterwongbbl, 2023-08-29:
Hi.   @ianderl 
The error message is not accurate. 
When an FTP transmission error occurs, There is a high probability that this information will be prompted. 
You can first check if the printer's network is working properly.
If the printer is too far away from WiFi, it will also prompt this error.

closed 2023-10-13 by walterwongbbl