bambulab/BambuStudio #2188: Unable to upload log from Mac or PC

fumeister, 2023-08-05:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac or Windows 11

### Printer

Bambu Lab P1S

### How to reproduce

https://bambulab.com/en/my/support/tickets/create

### Actual results

Would not let me upload any log compress or uncompress

### Expected results

Able to upload trouble log

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12268290/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2023-08-08:
@fumeister Are you requesting a feature?

fumeister, 2023-08-08:
Bug!

bambu123, 2023-08-09:
@sodachen22 用户说的是打印机的功能吗？Studio 上并没有上传日志的功能。