bambulab/BambuStudio #2190: Red colored filament is not matched in AMS slot.

Hologos, 2023-08-05:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

macOS 13.5 (22G74)

### Printer

Bambu Lab P1P + AMS

### How to reproduce

1. Open included project file.
2. Set filament color in AMS slot to red (see screenshots).
3. Set filament color to red (see screenshots).
4. Slice and click print.
5. See that red color is not correctly matched in AMS slot.

### Actual results

Red color is matched to whatever color in other slots. 

### Expected results

Red color correctly matched.

### Project file & Debug log uploads

[image]
[image]
[image]
[Combo case - middle part.3mf.zip](https://github.com/bambulab/BambuStudio/files/12268419/Combo.case.-.middle.part.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

Hologos, 2023-08-12:
Just did another project and it seems like all custom color are not matched correctly.

OzDruiD, 2023-12-04:
Following, sometimes see tis isssue as well.

SaltWei, 2024-01-18:
@MklBambu Please check whether this issue still exist. If not, please close the ticket.

Hologos, 2024-01-18:
@SaltWei I am using latest version and this issue still persists.

SaltWei, 2024-01-18:
> @SaltWei I am using latest version and this issue still persists.

Got it. @MklBambu Please follow this issue.

Hologos, 2024-01-18:
Here is a screen recording. Both slots are set to **Generic PLA** and the color is the same from the macOS color picker.

[image]

closed 2024-03-28 by MklBambu