bambulab/BambuStudio #2191: 优化悬垂切片走线逻辑 Optimize the overhang slice routing logic

wangkai294, 2023-08-06:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases, Built from source by yourself or others

### OS version

win10

### Printer

bambu lab p1p

### How to reproduce

[image]
[image]
走线逻辑没有老版本好，现在的走线逻辑无法实现无支撑打印
The routing logic is not as good as the old version, and the current routing logic cannot achieve printing without adding support

### Actual results

[image]
左边是老版本，右边是新版本
On the left is the old version, on the right is the new version

### Expected results

调整走线逻辑
Adjust the cable logic

### Project file & Debug log uploads

[对比.zip](https://github.com/bambulab/BambuStudio/files/12268573/default.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-08-07:
@QingZhangBambu 请最高优先级处理这个问题，hotfix版本发布后，一大波用户反馈overhang wall算的不对

QingZhangBambu, 2023-08-09:
the wall order and overhang detection had been fixed. you can get full functionality on version 1.7 hotfix2

closed 2023-08-09 by QingZhangBambu