bambulab/BambuStudio #2192: Filament selection on Prepare tab

JeffreyBrew, 2023-08-06:
**Is your feature request related to a problem? Please describe.**
The way filaments are displayed on the Prepare page of Bambu Studio can get confusing.  Sequentially number filaments is not a good idea.  Filaments should be numbered in the same manner as on the Device page.  In the picture titled "Filaments as shown on Prepare" there are nine filaments displayed.  Filaments 8 and 9 are actually loaded in D3 and D4, respectively.

**Describe the solution you'd like**
I'd like to see the representations of the AMS as shown in "Filaments as shown on Device" also displayed on the Prepare page of Bambu Studio.  When a user clicks on one of the AMS, the filament(s) loaded into that AMS are shown below numbered accordingly (A1, A2, A3, A4, B1, B2, B3, B4, C1, C2, C3, C4, D1, D2, D3, D4) as shown in "Proposed Filament selection for Prepare (A)" and "Proposed Filament selection for Prepare (B)".  All other parts of Bambu Studio (e.g., Flushing volumes, etc.) should also be updated to use the same filament numbering.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Filaments as shown on Prepare
[image]

Filaments as shown on Device
[image]

Proposed Filament selection for Prepare (A)
[image]

Proposed Filament selection for Prepare (B)
[image]

Panda631, 2023-08-09:
Good idea

Sonicboom247, 2024-01-15:
I agree, I was just searching for the same issue for the AMS lite for the A1. The AMS Lite is defined as 
1 and 4
2 and 3
[image]

whereas in Bambu Studio they are displayed as 
1 and 2
3 and 4
[image]
These need to be represented like for like so i am not constantly verifying slot placement.