bambulab/BambuStudio #2193: Bambu Cube V2 has a design mistake, there is an update to Bambu Cube V2R1

thrutheframe, 2023-08-06:
### Bambu Studio Version

1.7.2

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows 11

### Printer

Bambu Lab X1 series and P1 series

### How to reproduce

When "add Primitive" -> "Bambu Cube V2"
There is a mistake in the design of the cube. It is supposed to be supportless, but the design did not include the bridging needed. I have updated the 3mf which can be downloaded from: 

https://www.printables.com/model/222285-bambu-lab-xyz-calibration-cube/files

[image]

### Actual results

The square is printed in mid-air.
[image]

### Expected results

Layer37 has bridging support, follow by layer38 with the square so that it is not printed in mid-air

[image]

[image]

### Project file & Debug log uploads

no log file and I cannot add 3mf file here. so u need to download the new 3mf file from printables. thanks

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Merlwynd, 2023-08-06:
Do you know what folder in BS the cube stl or 3m file is in?…so we can download from Printables and replace it there….

DanBao-Bambu, 2023-08-07:
Thank you for your correction. We have replaced the new model you provided, but the name has not changed. We hope you are aware. Thank you again for your support.

DanBao-Bambu, 2023-08-08:
The version containing your updated files has not been released yet. If it is released, I will notify you. At that time, you can find your model in the "/resources/model/Bambu_CubeV2. stl" directory. I hope to answer your question.

Merlwynd, 2023-08-08:
> The version containing your updated files has not been released yet. If it is released, I will notify you. At that time, you can find your model in the "/resources/model/Bambu_CubeV2. stl" directory. I hope to answer your question.

Yes, thank you. The Bambu Cube V2R1 stl is posted on Printables and I guess I can download it and use it to replace the V2 version in the directory you provided…if I want to use it before you release the next version of Bambu Studio…

thrutheframe, 2023-08-08:
Updated cube V2R1 has been added to Bambu Studio V1.7.3. thank you very much and close.

DanBao-Bambu, 2023-08-08:
I am glad to see your reply and look forward to your valuable suggestions next time.