bambulab/BambuStudio #2195: Wall order is incorrect

Roussette5, 2023-08-06:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 13.5

### Additional system information

3,2 GHz Intel Core i7 6 cœurs
Intel UHD Graphics 630 1536 Mo
16 Go 2667 MHz DDR4

### Printer

P1S

### How to reproduce

1. Open STL in bambu studio
2. Slice with selecting wall order --> inside/outside/infill

[image]

### Actual results

When looking at the slice, it's doing : outside/inside/infill

[image]

### Expected results

It should be doing inside/outside/infill as requested

### Project file & Debug log uploads

[rotating-rings-toy.3mf.zip](https://github.com/bambulab/BambuStudio/files/12269017/rotating-rings-toy.3mf.zip)
[Archive.zip](https://github.com/bambulab/BambuStudio/files/12269018/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Yeello1, 2023-08-06:
It's only when "Arachne" wall is active

vjapolitzer, 2023-08-07:
I have observed this as well. Changing the seam setting between aligned and random may have an effect. On some steep ~60° overhangs, even with seam=aligned and walls=classic, the slicer is sometimes doing outside/inside/infill, which is negatively impacting the overhangs.

QingZhangBambu, 2023-08-09:
the wall order had been fixed, you can get full functionality on version 1.7 hotfix2

closed 2023-08-09 by QingZhangBambu

vjapolitzer, 2023-08-09:
@QingZhangBambu Nice work, thank you!  
I can verify that it is fixed for `Wall Generator = Arachne`, however I can still produce a sliced file with the wrong wall order with `Wall Generator = Classic`

See below example:  
[image]   
[debug_Tue_Aug_08_19_44_05_1699.log.0.zip](https://github.com/bambulab/BambuStudio/files/12297780/debug_Tue_Aug_08_19_44_05_1699.log.0.zip)
[WrongWallOrder8_8_23.3mf.zip](https://github.com/bambulab/BambuStudio/files/12297781/WrongWallOrder8_8_23.3mf.zip)

QingZhangBambu, 2023-08-09:
@vjapolitzer if you don't mind, you can open a new issue for that, and it is a corner case, a bit tricky to adjust it. But I will try.

Yeello1, 2023-08-09:
Thank you for the fast fix 😀

vjapolitzer, 2023-08-09:
> @vjapolitzer if you don't mind, you can open a new issue for that, and it is a corner case, a bit tricky to adjust it. But I will try.

@QingZhangBambu Okay, I have opened #2235 for this. Thank you!

QingZhangBambu, 2023-08-09:
@vjapolitzer OK, thank you.