bambulab/BambuStudio #2199: nozzle wipe during the print

bewgow, 2023-08-06:
**Is your feature request related to a problem? Please describe.**
I am printing mostly with PETG and the filament keeps oozing onto the nozzle and blob forms that eventually knocks off the print of the plate and print fails.

**Describe the solution you'd like**
I would like to see a feature where you could tell the printer to wipe the nozzle every 50 layers or so to keep the nozzle free of blobs.

**Describe alternatives you've considered**
I changed the temperature settings and extrusion and it is still a problem.

**Additional context**

[image]

mpetito, 2023-08-08:
Another variation of this that I think would be useful is if there was an on-screen utility command at the machine to wipe the nozzle.  Often times I can see there is a problem with filament buildup while a print is in progress, but this would be useful both when the machine is idle and while it is printing.  When used while printing, it could work as a sequence of pause, wipe, resume.

mOREsMod, 2023-12-13:
Can the camera/LiDAR see the nozzle? I imagine this can be automated. See buildup, wipe, go back to printing.

If not, an option to wipe every so many layers would be nice.