bambulab/BambuStudio #2201: Advanced: Order of inner wall/outer wall/infill

Ankerstjerne1982, 2023-08-06:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU: core i7-10700KF
Ram:32gb
SSD disk. 

### Printer

Bambu Lab X1-Carbon

### How to reproduce

Scale hook to 105%
Slice with standard settings for 0.4 nozzle and 0.20mm standard @bbl X1C 
preview 2nd layer, it does the outerwall before the inner wall, but the setting is set to inner/outer/infill
[image]

### Actual results

It makes the outer wall before the inner wall, and print does not stick to the model resulting in spaghetti failure

The advanced setting is inner/outer/infill

### Expected results

It should do inner wall before outerwall 

### Project file & Debug log uploads

[skadis_hook.zip](https://github.com/bambulab/BambuStudio/files/12270209/skadis_hook.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

SaltWei, 2023-08-07:
@QingZhangBambu Please check this issue ASAP!!!

QingZhangBambu, 2023-08-09:
The wall order had been fixed, you can get full functionality on version 1.7 hotfix2

closed 2023-08-09 by QingZhangBambu