bambulab/BambuStudio #2202: Ability to select a speed rating (Silent / Normal / Sport / Ludicrous) upon sending print job

k8stuff, 2023-08-06:
**Is your feature request related to a problem? Please describe.**
When printing a plate and there is a requirement to start the print with a different speed than normal, I need to start the job and essentially wait until all the calibrations are done before I can make sure the change to the speed is 'taken'. Until the calibration is finished it may not take the change from Normal to Silent for example so the best way is to wait about 7-10 minutes before changing the speed. A lot of time it works, but a lot of times it doesn't and it isn't guaranteed that a new setting is taken (i.e. sometimes it takes it, displays the wrong speed however, until the print starts).

**Describe the solution you'd like**
When printing a plate it would be great to not just have the ability to set Bed Leveling, Flow Dynamics Calibration, Timelapse and Enable AMS, but also a drop down with the required speed rating of Silent, Normal, Sport and Ludicrous. Bonus feature if the setting can be timed (i.e. Silent until 7am before it goes into Normal or Speed). Timed is not necessary and a really nice one to have, but the ability to make that selection is missing.

**Describe alternatives you've considered**
Only alternative is to change the speed setting straight in the slicer, but that would mean setting ever single speed if all I want is i.e. change the mode from Normal to Silent temporarily when kicking off a print (i.e. before bed time).

**Additional context**
See example in screenshot / mockup

[image]

kylek29, 2023-08-09:
I agree with this, I've felt the current method sorta obfuscates the speed feature to the point where I haven't used it since the first week with my printer (and I was a Kickstarter purchase, so I've had it for a long time).

The print send dialog is a good option.

randyre, 2023-09-07:
+1 on this. Should have been there from the beginning.  Would also like to see a print at X speed for Y layers before going to high speed.

GuzziRaz, 2023-09-15:
I too would like to see this. While at it, an additional “quality” speed level for, say, 90% (or whatever you see fit) would be nice.