bambulab/BambuStudio #2203: No "+" button in height range modofier in Hungarian language

AndrewTdi, 2023-08-06:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 x64

### Printer

X1C+AMS

### How to reproduce

I use the slicer in Hungarian language and there is no + button on the height Range modifier tab. If I set the slicer language to English, I can find it right away.

### Actual results

[image]

### Expected results

I should see the "+"  button at the height range modifier tab in Hungarian language too

### Project file & Debug log uploads

[gomb.zip](https://github.com/bambulab/BambuStudio/files/12270248/gomb.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12270249/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-08-07:
@bambu123 应该是匈牙利语言太长了，顶出去了，帮忙分配一下内部哪位同事帮忙修一下。1.8版本修

bambu123, 2023-08-07:
@LiZ-Li-BBL help

LiZ-Li-BBL, 2023-10-24:
Hello,
We've fixed this issue, the button will display normally in the next release.

closed 2024-01-16 by LiZ-Li-BBL