bambulab/BambuStudio #2204: resonance calibration, skipping steps when hotend moves into position

sharps89, 2023-08-06:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab github releases

### OS version

windows 11

### Printer

Bamu labs p1s

### How to reproduce

goto device, click calibrate, then resonance calibration. 
after the nozzle clean it moves the toolhead into the final posistion but it skips steps with a loud screeching sound.
then after the resonance test is complete the too head tries to move and hits the end on the travel and makes a loud noise.

### Actual results

after the nozzle clean it moves the toolhead into the final posistion but it skips steps with a loud screeching sound.
then after the resonance test is complete the too head tries to move and hits the end on the travel and makes a loud noise.

### Expected results

after nozzle clean, toolhead should move to centre of the bed to do test but it skipps steps and stays top left of bed. 

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12270296/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

allthemoist, 2023-08-06:
Can confirm mine is also doing this!

allthemoist, 2023-08-07:
Just to confirm this only happens when starting the calibration via bambu studio and bambu handy. If Calibration is started via the menu in the printer screen the calibration operates as it should.