bambulab/BambuStudio #2205: Printing performance regression Studio 1.7.2.51 compared to models sliced with 1.6

NetOneGames, 2023-08-06:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

OS Name	Microsoft Windows 10 Pro
Version	10.0.19045 Build 19045
System Model	HP Z420 Workstation
Processor	Intel(R) Xeon(R) CPU E5-1650 0 @ 3.20GHz, 3201 Mhz, 6 Core(s), 12 Logical Processor(s)
Installed Physical Memory (RAM)	32.0 GB
Adapter Type	NVIDIA GeForce GTX 1080, NVIDIA compatible

### Printer

P1P

### How to reproduce

Finding print performance regression with new 1.7 Studio update

Please see
https://forum.bambulab.com/t/print-quality-degrade-after-studio-1-7-update/23209

Slicing same model in 1.6 vs 1.7.   1.7 is printing badly.  

I am sorry I cannot upload problem model as it is customer IP
I am just now creating similar test model and running some tests I will upload models, results  shortly thanks.

### Actual results

TBA. Will be updated shortly.

### Expected results

TBA. Same performance as 1.7

### Project file & Debug log uploads

[TBD.zip](https://github.com/bambulab/BambuStudio/files/12270391/TBD.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SaltWei, 2023-08-07:
This is bug caused by wall generator and wall printing order. There has a bug at latest version. The wall order is absolutely wrong at arachne wall generator.

SaltWei, 2023-08-07:
We are fixing ASAP

faroutride, 2023-08-07:
When can we expect the fix?
This is kind of a big deal. 
Thanks.

NetOneGames, 2023-08-07:
cheers guys. I won't bother update the report then.

If I could make one related request.
 it seems that  when updating studio you are overriding user custom print setting with new default setting.  I.e. Classic vs arachne
Please do not do this. 
User custom setting should be sacred.

Hologos, 2023-08-07:
They are not overwriting anything, it's the way the profiles are derived. It inherits everything except what you change.

NetOneGames, 2023-08-07:
Once a user makes a custom print profile all settings in that profile, even unchanged settings, must be locked in unless changed by user.

Hologos, 2023-08-07:
That's not how it works in Bambu Studio, though. You have to keep in mind that Bambu Lab printers are targeting new audience. The ones that have zero experience with 3D printing. Once an unexperienced user would create his own profile, he would lose out on all updates and tweaks to profiles Bambu Lab brings. He could have problems printing and contact their support. That is the reason Bambu Lab is doing it this way.

If you know how it works, you can act accordingly. Until Bambu Lab change their mind (which I don't believe they will for reasons I described), it will work this way.

NetOneGames, 2023-08-07:
Disagree with every point you make above , in fact you are plainly wrong on most points however this is not a forum.  And I'm sure u are aware of the irony that it was the fact that my custom profile was changed without my knowledge that led to the problem being a problem. 

Custom profiles are created for very specific reasons , they have usually been fine tuned over many iterations and can become extremely valuable and important assets for some users like myself.  Notices of incompatibilities  or optional upgrades can be communicated but profiles should never be messed with.  Ever.

faroutride, 2023-08-07:
Can we keep this on topic (wall order messed up with Arachne Engine)?
The argument you both have is relevant though, so maybe open a different ticket for this (profile change with updates)?

Cheeers

SaltWei, 2023-08-08:
> Can we keep this on topic (wall order messed up with Arachne Engine)?

This is bug and we are handling. Will fix it ASAP

Hologos, 2023-08-08:
> Disagree with every point you make above , in fact you are plainly wrong on most points however this is not a forum. And I'm sure u are aware of the irony that it was the fact that my custom profile was changed without my knowledge that led to the problem being a problem.

You can disagree as much as you want, I am just stating the facts and how it's implemented. I am not saying it's the correct approach.

> Custom profiles are created for very specific reasons , they have usually been fine tuned over many iterations and can become extremely valuable and important assets for some users like myself. [...]

And that is why I wrote you how it works currently. So you know and act accordingly. You can either fix your profiles in the meantime until Bambu changes the behavior or deriving profiles or you can continue to put words in my mouth I haven't said. 🤷‍♂️

As others have said, this is not the place. This is my last response about this topic. Cheers ✌️

QingZhangBambu, 2023-08-09:
the wall order had been fixed, you can get full functionality on version 1.7 hotfix2

closed 2023-08-09 by QingZhangBambu

NetOneGames, 2023-08-09:
Awesome 👍