bambulab/BambuStudio #2206: Touble with SD cards

fishinone, 2023-08-07:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

BambuLab X1C

### How to reproduce

Try to use the SD card

### Actual results

Varied problems. Sometimes it says the new SanDisk micro SD card isn't there, other times it says that it's not formatted, then it will say there is no network. These things happen using the printer screen, using Bambu Stidio and using Bambu Handy.

### Expected results

Let's fix it.

### Project file & Debug log uploads

[log.zip](https://github.com/bambulab/BambuStudio/files/12271561/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

tsmith35, 2023-08-07:
Seeing similar. With X1C and Studio (both latest available), choosing Device > MicroSD Card > Model initially comes up showing previews for first 5 items and generic "Bambu Lab" boxes for last 5 items. A couple seconds later, Studio crashes without warning.

bambu123, 2023-08-08:
> Seeing similar. With X1C and Studio (both latest available), choosing Device > MicroSD Card > Model initially comes up showing previews for first 5 items and generic "Bambu Lab" boxes for last 5 items. A couple seconds later, Studio crashes without warning.

Please send me the logs include the .dmp files.

bambu123, 2023-08-08:
It seams the network is bad, and no file list is received. But the liveview is ok. 
So that maybe a printer's bug. @sodachen22

bambu123, 2023-08-08:
@fishinone can you see timelapse files from studio?

sodachen22, 2023-08-08:
1. Regarding SD card, "Sometimes it says the new SanDisk micro SD card isn't there, other times it says that it's not formatted", is the SD card status on the printer screen is the same as on the Studio? If SD card status on the screen is the same as on Studio, please check the FPC cable of the screen to make a better connection between the SD card slot and the AP board.
2. Regarding network problem, if the WiFi signal from the router is not good enough, please have a test with a hotspot created by the mobile phone.

fishinone, 2023-08-08:
Yes

I re-seated the display connector yesterday. Haven't had any problems after
that, at least so far.

bambu123, 2023-08-09:
@tsmith35 
From the logs you provided, I found out that it was because of the third-party 3mf file parsing Crash. We will fix it ASAP. Before we release a new version, in order to use this feature, it is recommended to manually delete the non-BambuStudio saved files in the SD card first.

fishinone, 2023-08-10:
Well the problem reoccured.
Is there something wrong with the Bambu formating code?
Again, All files on the card were from Bambu Labs!

bambu123, 2023-08-10:
> Well the problem reoccured. Is there something wrong with the Bambu formating code? Again, All files on the card were from Bambu Labs!

Which files do you recently add to SD card？ Where do they come from？

fishinone, 2023-08-10:
I slice my models in Bambu Studio and use Send to add them to the SD card.
The SD card seems to become corrupted sometimes while it's printing.
Sometimes files seem to send but they are corrupted as soon as I get to the
printer.

fishinone, 2023-08-11:
[debug_network_Thu_Aug_10_12_11_49.log](https://github.com/bambulab/BambuStudio/files/12323904/debug_network_Thu_Aug_10_12_11_49.log)
[debug_network_Thu_Aug_10_12_01_37.log](https://github.com/bambulab/BambuStudio/files/12323905/debug_network_Thu_Aug_10_12_01_37.log)
[debug_network_Thu_Aug_10_11_46_14.log](https://github.com/bambulab/BambuStudio/files/12323906/debug_network_Thu_Aug_10_11_46_14.log)
[debug_network_Thu_Aug_10_11_24_11.log](https://github.com/bambulab/BambuStudio/files/12323907/debug_network_Thu_Aug_10_11_24_11.log)
[debug_network_Thu_Aug_10_11_22_57.log](https://github.com/bambulab/BambuStudio/files/12323908/debug_network_Thu_Aug_10_11_22_57.log)
[debug_network_Thu_Aug_10_08_03_
[debug_network_Fri_Aug_11_10_28_52.log](https://github.com/bambulab/BambuStudio/files/12323913/debug_network_Fri_Aug_11_10_28_52.log)
24.log](https://github.com/bambulab/BambuStudio/files/12323909/debug_network_Thu_Aug_10_08_03_24.log)

Her you go. Yesterdays los and todays.

bambu123, 2023-08-14:
> [debug_network_Thu_Aug_10_12_11_49.log](https://github.com/bambulab/BambuStudio/files/12323904/debug_network_Thu_Aug_10_12_11_49.log) [...]

Thanks for your reply, but these logs are not enough, especially the missing .DMP file.

fishinone, 2023-08-14:
So, state precisely what you want and where to get it.

Ron Davis
RC&C Design
(661) 753-6149

CONFIDENTIALITY NOTICE: The information contained in this electronic mail
and its attachments are privileged and confidential information and are
intended for the use of the individual(s) or entity named above and those
who have been specifically authorized to receive it. If you are not the
intended recipient, you are hereby notified that any use, copying or
distribution of this communication is strictly prohibited.

bambu123, 2023-08-14:
> So, state precisely what you want and where to get it. 

Please simply zip all files in the folder %APPDATA%\BambuStudio\log and send the zip to me.

fishinone, 2023-08-14:
Here you go.

Ron Davis
RC&C Design
(661) 753-6149

CONFIDENTIALITY NOTICE: The information contained in this electronic mail
and its attachments are privileged and confidential information and are
intended for the use of the individual(s) or entity named above and those
who have been specifically authorized to receive it. If you are not the
intended recipient, you are hereby notified that any use, copying or
distribution of this communication is strictly prohibited.

bambu123, 2023-08-15:
> Here you go. Ron Davis RC&C Design (661) 753-6149 CONFIDENTIALITY NOTICE: The information contained in this electronic mail and its attachments are privileged and confidential information and are [...]

 Haven't seen any attachments.

fishinone, 2023-08-15:
Aparently, you did not recieve my zip file via email. I made another one today.

I actually expect an answer to the sandbagging question. Do I need to request warranty replacement now before you run out the clock?

[log.zip](https://github.com/bambulab/BambuStudio/files/12353709/log.zip)

bambu123, 2023-08-16:
> Aparently, you did not recieve my zip file via email. I made another one today.
> 
> I actually expect an answer to the sandbagging question. Do I need to request warranty replacement now before you run out the clock?
> 
> [log.zip](https://github.com/bambulab/BambuStudio/files/12353709/log.zip)

@sodachen22 will answer your another question.

fishinone, 2023-08-16:
and who is going to look into the bad file handling?

fishinone, 2023-08-16:
When Bambu123 states

"From the logs you provided, I found out that it was because of the third-party 3mf file parsing Crash. We will fix it ASAP. Before we release a new version, in order to use this feature, it is recommended to manually delete the non-BambuStudio saved files in the SD card first."

What third party file or files specifically is being refered to? The SD card was bought new and formatted with the printer. At that time it had never been used in another device.

sodachen22, 2023-08-18:
@fishinone Could you please provide more information about the data corrupted of the SD card? And there has been problem with SD card which was resolved after a re-seated of FPC cable? If the problem is from hardware, you may need create a ticket according to https://wiki.bambulab.com/en/create-a-support-ticket.

fishinone, 2023-08-18:
I don't know if I have any more information.
I format the cards in the X1C they work for a while then they quit working.
This happened with old 2Gb cards that I've had and used for a long time
with no problems then with a new 512Gb card that I bought for the X1C. I
reformat them and they work ok for a while then quit again.
It's been working lately although I'm certainly not confident in it.
I'm tired of losing jobs that are already figured out. Now, with software
changes they don't come out exactly like the ones that worked before.
I already researched the flex cable for the screen. If it worked, it wasn't
100 percent.

fishinone, 2023-08-18:
I'm unhappy with the casting about that I've heard from Bambu regarding
this. Some off the replies seen to be looking for a scapegoat. I don't care
who's to blame. I just want it to work.