bambulab/BambuStudio #2208: Imported config doesn't show up in the drop down menu

sadboybushido, 2023-08-07:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS Ventura 12.0.1

### Additional system information

APPLE M2 16GB

### Printer

BambuLab P1S

### How to reproduce

1. File
2. Import > Import configs...
3.  Choose file and click ok

### Actual results

It keeps saying that the preset exists already and the preset selection menu flickers for a split second after clicking Yes. But when I click to choose the preset, it never shows up. Only the system presets. No user presets. However, if I log out of my profile and import the config, it shows up. Then again, if I log in again, it disappears.
I also tried logging in and out, and closing the app, reinstalling it. Nothing has helped.

### Expected results

It is supposed to be there.

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-08-08:
Maybe the preset is not compatible with the printer preset you have selected, so it is there but not visible.

sadboybushido, 2023-08-08:
> Maybe the preset is not compatible with the printer preset you have selected, so it is there but not visible.

No, everything is compatible. Just migrating preset from one PC to another with the same Bambu Studio version. 
Besides, as I said, the preset does appear and can be used if I am not logged in my account. But also not every single time.

bambu123, 2023-08-08:
Not sure what the problem is. If you want to synchronize presets between multiple PCs, please log in to your account and enable automatic synchronization of user presets. If you still want to synchronize manually, it is best to manually delete the current presets of the user directory first (some previous presets may be problematic). Users log in and out, and the program will use data from different preset directories.

To confirm if it is a compatibility problem, you can check whether the files in the preset directory contain the presets you just imported after importing user presets. If they do, the presets are incompatible, so they will not be visible in the selection box. You can also send us the preset files that were unsuccessfully imported for analysis.

sadboybushido, 2023-08-15:
Hello. I was able to understand why. The profile disappears in Orca slicer too.
After I change the printer profile from P1S to P1P the preset appeared.
It looks like if you choose the P1S as your printer, it will show X1C presets only.
Can you guys fix it?

bambu123, 2023-08-16:
> Hello. I was able to understand why. The profile disappears in Orca slicer too. After I change the printer profile from P1S to P1P the preset appeared. It looks like if you choose the P1S as your printer, it will show X1C presets only. Can you guys fix it?

Yes. User-defined presets are generally only compatible with one model, so you need to create custom presets based on the presets of the correct model.

richard-scott, 2023-08-24:
Say you have a filament profile made on X1C, and you want to use that on a P1S
Is there any way to import (or copy) presets to a new model?

b-g, 2023-11-25:
Same problem here. Would be great to get this fixed finally. @bambu123 any updates?

bambu123, 2023-11-27:
> Say you have a filament profile made on X1C, and you want to use that on a P1S Is there any way to import (or copy) presets to a new model?

Sorry for the delay. The answer is no. We still recommend creating user-defined presets for different printer models.

DanBao-Bambu, 2024-01-18:
@sadboybushido 
> It looks like if you choose the P1S as your printer, it will show X1C presets only.

I know what you're talking about, but in fact, it's not a mistake because when designing presets, the same Filament preset can actually be used by multiple machines. As you said, some of the presets of the X1C printer can be used by P1S, so this phenomenon may occur.
I'm glad to answer your questions. Do the questions I mentioned before still exist?