bambulab/BambuStudio #2209: "Reload from Disk" stopped working after upgrade to 1.7.2.51

sophof, 2023-08-07:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

1. Create a 3mf file
2. Load it in bambu studio and slice it
3. Change file in editor and overwrite old 3mf file
4. Click "reload from disk"

### Actual results

Nothing will happen, there will also be no errors.

### Expected results

File is reloaded and the model is exchanged with the latest version (was working fine before).

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2023-08-08:
If you have modified a 3mf file that has been opened as a project, you should reopen it, not reload from disk.

bambu123, 2023-08-09:
> As I said, it worked before and the 3mf file is not a project, it is simply the output format for an object these days if you use fusion 360. It would be a weird regression if this suddenly isn't possible anymore.

@DanBao-Bambu will check that.

DanBao-Bambu, 2023-08-22:
Please check if the gif below meets your expectations.

[image]

sophof, 2023-08-23:
It does to an extent. The difference is that I save a new model under the same name from fusion 360. This is saved as a 3mf file, that is the default option. I just tried it with a new model and suddenly it worked again. There's been a small update in the meantime (but nothing about this was in the release notes), so maybe it was just a fluke?

DanBao-Bambu, 2023-08-23:
I understand. I will verify the scenario you mentioned and confirm that my modifications to the code can meet your expectations.

closed 2024-01-18 by DanBao-Bambu