bambulab/BambuStudio #2210: [Cancel] button on 'Edit Plate Name' is not working

Lemonadel, 2023-08-07:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Catalina 10.15.7 (19H15)

### Printer

Bambu Lab P1P

### How to reproduce

1. Right click on plate -> Edit Plate Name -> Type "Test123" as a name ;)
2. Click [OK] button to save new name
3. Again do the step #1 but type "Test1" (i.e. remove "23" from string end)
4. Click [Cancel] button to NOT save newly typed Plate name

### Actual results

Step #4 saves "Test1" instead of "Test123" ( previous Plate name)

### Expected results

Previous Plate Name is kept / saved - so it's "Test123" as in Step #1

### Project file & Debug log uploads

[BugReport.zip](https://github.com/bambulab/BambuStudio/files/12280830/BugReport.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2023-08-08:
Thank you for your discovery. It's my problem. I have already found the problem. Thank you~

Haidiye00, 2023-08-08:
This issue will be fixed in version 1.8

closed 2024-01-16 by SaltWei