bambulab/BambuStudio #2213: 2nd AMS with Generic Filament

DGable, 2023-08-07:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 Home

### Printer

X1C Carbon

### How to reproduce

1. Open Bambu Studio
2. Select Device
3. Select 2nd AMS Icon
4. Slots B1 & B4 Showing Filament is editable
5. Slots B2 & B3 indicating a RFID Tag present
6. B1 thru B4 loaded with Generic PLA
[image]

### Actual results

Slots B2 & B3 information is not editable. slots B2 & B3 are indicating a RFID tag in use, even though only generic PLA is loadedin slots B1, B2, B3, & B4. 

### Expected results

Should be able to edit each slot indicating what filament is in that particular slot

### Project file & Debug log uploads

[BambuNetworkEngine.zip](https://github.com/bambulab/BambuStudio/files/12281427/BambuNetworkEngine.zip)
[PN Sealing Spool.zip](https://github.com/bambulab/BambuStudio/files/12281432/PN.Sealing.Spool.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MagicMilamber, 2023-08-07:
exactly the same Setup und the same Problem

QingZhangBambu, 2023-08-08:
I am not quite sure what your problem is, for example, do you mean that should show Bambu PLA basic instead of PLA?
[image]

QingZhangBambu, 2023-08-08:
we show the filament name on prepare page, the device page only shows the generic category
[image]

QingZhangBambu, 2023-08-08:
Closed for no responding

closed 2023-08-08 by QingZhangBambu

SaltWei, 2023-08-09:
Why close so rapidly?