bambulab/BambuStudio #2217: Speed displayed does not match between Windows app and printer display

AuryF, 2023-08-08:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro 22H2

### Printer

Bambu X1 Carbon w/ AMS

### How to reproduce

Open two copies of Studio, each with a project to print.
First Studio copy - Start print of project A with speed already at 100%
First Studio copy - Set speed via Studio to 166%
When print finishes, Studio copy A and printer should show 166%
Second Studio copy - Start print of project B
Second Studio copy - Set print speed to 124%

At this point the printer should now be showing 124% but it instead shows 166% still. 

### Actual results

Both copies of Studio show print speed at 124% but the printer shows 166%. 

### Expected results

All copies of Studio and the printer should show the same speeds.

Attached the most recent log in the folder but it doesn't look to be from when this happened.

### Project file & Debug log uploads

[Bambu Roomba Bumper.zip](https://github.com/bambulab/BambuStudio/files/12286514/Bambu.Roomba.Bumper.zip)
[debug_Mon_Aug_07_17_08_08_50064.log.zip](https://github.com/bambulab/BambuStudio/files/12286525/debug_Mon_Aug_07_17_08_08_50064.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

bam-winterfell, 2023-08-29:
@AuryF Please provide a screenshot of the bug when it occurred. Thank you

bam-winterfell, 2023-12-21:
If you have any new findings regarding this issue, please create a new one. thank you.

closed 2023-12-21 by bam-winterfell