bambulab/BambuStudio #2222: Manual Flow Dynamics Calibration Pattern Removal Difficulty / Improvements

mpetito, 2023-08-08:
**Is your feature request related to a problem? Please describe.**
The manual flow dynamics calibration feature prints a label for each line containing several digits. The label text is not attached to the line, and when removed from the bed with a scraper, often results in several tiny plastic digits flying everywhere. It makes a mess and is not really necessary.

**Describe the solution you'd like**
There are several simple alternatives or improvements that could be made:

1. Reduce the number of digits / remove the decimal place.  Instead of 0.050 you could easily have 50 and inform the user to divide by 1000 (or do it automatically in the UI).
2. Provide an option to skip printing the digits entirely.  I can count the number of lines from the bottom and multiply just as easily.
3. In the first layer, print a rectangular raft for the digits area of each line, then go back in a second layer and print the digits over the raft.  This way the digits are all connected to something a bit easier to remove.

**Describe alternatives you've considered**
Using an alternative PA calibration pattern such as the [Ellis's Pattern Method](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html).

**Additional context**
I needed to run a manual calibration for a transparent PETG filament and ended up with lots of very hard to see plastic digits all over the place.

SaltWei, 2023-08-09:
> Using an alternative PA calibration pattern such as the [Ellis's Pattern Method](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html).

Good idea! But can this be freely used in BambuStudio? Maybe we need the permission from the designer.

mpetito, 2023-08-09:
It's always good to seek permission, and I do see that this has also been incorporated into the latest Orca Slicer beta.

https://github.com/SoftFever/OrcaSlicer/pull/1547

thewildmage, 2023-08-09:
> But can this be freely used in BambuStudio? Maybe we need the permission from the designer.

I put in the PR to add Ellis' Pattern Method to Orca Slicer and looked into the licensing when I was working doing so. What I found was that the [instructions for how to read the results](https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/pattern_method.html) and all of [Ellis' guide] don't have a license and so can't be used freely. (There's an [open Issue asking about licensing](https://github.com/AndrewEllis93/Print-Tuning-Guide/issues/83) but no response thus far.)

The [pattern generator itself](https://github.com/AndrewEllis93/Pressure_Linear_Advance_Tool) though is licensed as GPL-3.0. In fact, it's a modified version of the [Marlin generator](https://marlinfw.org/tools/lin_advance/k-factor.html), created for the [Marlin documentation](https://github.com/MarlinFirmware/MarlinDocumentation) by [Sineos](https://github.com/Sineos/k-factorjs), which is all, of course, GPL-3.0 licensed as well.