bambulab/BambuStudio #2225: AMS forgetting K Value (and management of multiple printers)

DamoclesSword, 2023-08-08:
### Bambu Studio Version

1.7.2.51

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10 

### Printer

Bambu Lab X1 Carbon

### How to reproduce

As a prerequisite to this, I am assuming you already have calibrations that have been generated and saved to an AMS. 

There are two bugs that occur here. The first must be solved so that the 2nd one can be shown. 

**The First Bug**

1. Open Bambu Studio
2. In the Device Tab, select a 2nd printer.
3. In most cases, all the K Values are now zero
4. Click the edit option on a filament
5. Click the Drop Down Menu for the PA Profile

Though my mouse is not shown due to snipping an image, you can see the box is green where I have clicked the drop down menu and nothing has happened. (this not just me hovering over it, the option has been clicked. You can tell because the grey portion of the background has a slight green tint which is not present when just hovering over it)
[image]

In most cases, this menu is bugged and will not allow for a value to be selected. 

Windows registers the menu is open as if the menu is visible, though the menu never appears, you cannot click away until the menu is closed. 

To resolve this:

1. Close the menu
2. Go the the Calibration Page,
3. Select Manage Results 
Here, you can see I clearly have values generated that were previously saved to the AMS (more on this in a moment) 
[image]
4. Return to the Device Tab
5. Now, the AMS will have K Values and they will be editable
[image]

With this initial bug solved, a new one occurs in other scenarios. 

**BUG TWO**
Note: The same printer is still selected

1. Ensure that the AMS has the correct selected K value for each filament in the Device Tab 
2. Slice a print and Select which printer to send the print to. 
3. If the printer is not the printer that is selected in the device tab, at times, the K Value zero out on the printer the file was sent to. This is not everytime and potentially is affected by the printer being inactive for awhile. Or, it could be the same issue as the first bug. 

The error seems to be dependent on what printer is selected in the Device Tab. The error is not a graphical one as I can see print quality differences on prints sent to a printer with the values at zero (that were previously selected) and ones where the print is stopped and the values are reselected. 

As a potential extra bug, this is all exacerbated if you multiple instances of Bambu Studio open. Due to my use case, that is a requirement and you can issues where in one instance, the AMS has K Values and in another, they are zero. 

### Actual results

[image]

Above, you can see that for this printer, all the AMS values are zero. While there can be a delay when you first open Bambu Studio for the AMS to show a K value, this is not related to that. 

### Expected results

For the first bug:

- the drop down menu should be selectable. However, once a K Value is selected, the AMS/Bambu Studio should not forget it. 

For the second bug:

- the AMS should not forget the K Value

For the Third _kinda_ bug:

- K Value should be stored on the printer such that even with multiple instances of Bambu Studio open, they all still know which filaments are in a AMS, so too should this be true with the K Value. It would also be nice to manually be able to input a K Value and edit the filament type. 

### Project file & Debug log uploads

[Save for Github_Second Bug Report.zip](https://github.com/bambulab/BambuStudio/files/12294161/Save.for.Github_Second.Bug.Report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

DamoclesSword, 2023-08-16:
Given that this issue is still open,  I am assuming this was not fixed with the latest 1.7.4 release? Could it also be confirmed whether or not this is a UI bug, or an actual software issue. 

By that I mean: 

assuming values were previously saved to the AMS, when the AMS shows "0.00" for the K value, but will display it once the "manage results" workaround is used, does that mean the 0.00 is what the printer would have used if a print was sent? Or, would the printer have used the saved K Value, but the UI is just displaying 0.00.

DamoclesSword, 2023-08-18:
I can confirm with the latest version of Bambu Studio that this bug is still very much present.

ncr, 2023-09-26:
Same here, we have a few carbons and this happens every day. A fix would be greatly appreciated!

DilapidatedMeow, 2024-09-12:
I am still experiencing this in 2024!

Tiberiumxx, 2025-01-18:
same problem with version 1.10, please fix it

DilapidatedMeow, 2025-01-18:
I've come to the conclusion that this is actually just a visual problem and the printer is actually using the custom value

If you print something with the calibrated values then delete it and force it to use default (0.020) the difference is striking - most obvious with filaments further away from 0.020, obviously)

Tiberiumxx, 2025-01-18:
> I've come to the conclusion that this is actually just a visual problem and the printer is actually using the custom value in the current version
> 
> If you print something with the calibrated values then delete it and force it to use default (0.020) the difference is striking - most obvious with filaments further away from 0.020, obviously)

Really is not only visual, i see the defect in print,when i replace spools for new one return to default pa,  now i just testing with add m900 in gcode print start for be sure . i need to multi print same object and need to be perfect. i also send a ticket to bambulab for this problem, also i have multi print x1 and p1s and i think is better make this setting to material profile (similar to orca) .
With some fiber filamente pa need more tuning and 0.02 standard to 0.04 calibrated can make huge differance for me.