bambulab/BambuStudio #2227: Error in Dutch translation

Pedro-Newbie, 2023-08-08:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

In de Dutch file (BambuStudio_nl.po) on line 1421 is the: msgid "OK" this is translated on line 1422 as: msgstr "Offline"
This must be also OK
[image]

### Actual results

[image]

### Expected results

Please revert the translation to OK

### Project file & Debug log uploads

n.a.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DanBao-Bambu, 2023-10-24:
Hello, the issue has been resolved. Thank you for your question. If you have any other questions, please feel free to raise them.

Pedro-Newbie, 2023-10-31:
The issue still remains, the aren’t any changes made in the last 2 months.

DanBao-Bambu, 2024-01-18:
We translate "OK" into "Bevestigen" in Dutch. As shown in the figure:
[image]