bambulab/BambuStudio #2229: [ISSUE] LAN camera liveview is not a "live" view. It has long delays

BambuLabDiscord, 2023-08-08:
In LAN Only mode, the camera liveview is not "live". It has significant delays. What the camera shows is behind what actually happens. In a print that last some 17 minutes, the delay is more than 8 minutes at the end, i.e. the camera view still shows the printing going more than 8 minutes after the job status is 100% complete. 

A "live" view is needed in order to effectively monitor the printing. The only remedy right now is to constantly stop and reload the camera to reduce the lag in the liveview, which is not convenient.  

Running on Bambu Studio version 1.7.3.50, Firmware version  01.06.01. Both are the latest as of today.

bambu123, 2023-08-10:
@BambuLabDiscord 
Please send me the studio logs.

bambu123, 2024-02-06:
I have fixed a similar issue on the macOS platform. Is the new version already fixed?

closed 2024-02-06 by bambu123