bambulab/BambuStudio #2230: Video Laggy and slow to connect going through Bambu Servers

TheCableGuy99, 2023-08-08:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11 Pro

### Additional system information

Upgraded MSI GS75 10SE Gaming Laptop
CPU: Intel i7-10750H @ 2.60GHz
RAM: 32GB
GPU: nVidia GeForce RTX 2060
NVMe: Samsung SSD 970 EVO 2TB

### Printer

Bambu Lab X1C (I have 3 and all 3 are affected)

### How to reproduce

1) Try to monitor the live view camera when in normal mode (using Bambu Servers) when printing or not

### Actual results

It's slow to connect and very laggy and delayed (a screenshot will not show this).

### Expected results

I expect the video to be fairly smooth even if there is a little delay but it's practically unwatchable.

I disabled all other network activity and it's still the same. So I switched to LAN mode and as soon as I use LAN mode the video is fast to connect and buttery smooth.

This leaves the issue with either my internet connection, route to your servers or your servers being congested.

My internet downstream is 900Mbps download and 110Mbps upload and is rock solid. It's a business line and using very little. bandwidth. Speedtests show around 920Mbps down and 108Mbps upload showing my internet speeds are more than fast enough to video speeds.
Speedtest taken just now: https://www.speedtest.net/result/15095913802.png

The above shows it's not my connection. I also highly doubt it's any routing to your servers. I really think your European servers are over congested. 

This has been going on some time is seems to be getting worse. It wasn't like this initially when I bought the printers.

I have also formatted my PC and tried with other machines. The only thing that resolves it is going to LAN mode.

It's the same with many previous versions of Bambu Studio and ANY project I print.

### Project file & Debug log uploads

[debug_network_Tue_Aug_08_18_42_35.log](https://github.com/bambulab/BambuStudio/files/12296873/debug_network_Tue_Aug_08_18_42_35.log)

Pick any project file, it really doesn't matter, it's been happening with ALL projects for months. I've only been able to realise it's the servers at fault recently since I can now view the video in LAN mode and it works.

### Checklist of files to include

- [X] Log file
- [X] Project file

bambu123, 2023-08-09:
Are you connecting to the P1P printer?  The P1P printer pushes liveview with 1fps, that is one picture a second.

TheCableGuy99, 2023-08-09:
> Are you connecting to the P1P printer?  The P1P printer pushes liveview with 1fps, that is one picture a second.

No these are all X1C's and it's the same on all of them. Rubbish through the cloud and absolutely fine and smooth in LAN mode.

bambu123, 2023-08-09:
> Reference i

OK. Please send all log files to me. The above one is not enough.

TheCableGuy99, 2023-08-09:
This is the whole log directory.

Thanks :)

[log.zip](https://github.com/bambulab/BambuStudio/files/12297412/log.zip)

TackleBerryCrash, 2023-08-09:
I second the existence of the issue on my end too. Speed and frame rate of the streaming service has been going down hill last couple months. X1C or P1P regardless I can not get more than 1-2 fps now if I see any.
I had to check progress just today on my prints today from the mobile app, I could not event pass the spinner today with a solid connection.

bambu123, 2023-08-09:
> I second the existence of the issue on my end too. Speed and frame rate of the streaming service has been going down hill last couple months. I had to check progress just today on my prints from the mobile app. I could not even pass the spinner today with a solid connection.

I found you are using LAN Only mode live view. Will the live view be ok in LAN Only mode?
You can turn off LAN Only mode from printer screen.

TackleBerryCrash, 2023-08-09:
> I found you are using LAN Only mode live view. Will the live view be ok in LAN Only mode? You can turn off LAN Only mode from printer screen.

I am not using Lan Only Mode, maybe OP is

TheCableGuy99, 2023-08-09:
> > > I second the existence of the issue on my end too. Speed and frame rate of the streaming service has been going down hill last couple months. I had to check progress just today on my prints from the mobile app. I could not even pass the spinner today with a solid connection.
> > 
> > 
> > I found you are using LAN Only mode live view. Will the live view be ok in LAN Only mode? You can turn off LAN Only mode from printer screen.
> 
> I am not using Lan Only Mode, maybe OP is LAN ONLY Means I can not use the mobile app but I use the mobile app extensively

I believe he's asking if you can put it in LAN only mode and try viewing through Bambu Studio to see if it resolves the issue like it does for me.

TackleBerryCrash, 2023-08-09:
> I believe he's asking if you can put it in LAN only mode and try viewing through Bambu Studio to see if it resolves the issue like it does for me.

I dont think so

TackleBerryCrash, 2023-08-09:
Also putting them into LAN only mode defeats the purpose of having cloud capable printers, counter productive...
It will work likes yours did but I am not gonna use the LAN only mode ever so cloud streaming should just work as is.

bambu123, 2023-08-09:
> This is the whole log directory.
> 
> Thanks :)
> 
> [log.zip](https://github.com/bambulab/BambuStudio/files/12297412/log.zip)

Sorry. I'm looking into this log. LAN only mode is on in this log.

TackleBerryCrash, 2023-08-09:
> > This is the whole log directory.
> > Thanks :)
> > [log.zip](https://github.com/bambulab/BambuStudio/files/12297412/log.zip)
> 
> Sorry. I'm looking into this log. LAN only mode is on in this log.

@TheCableGuy99  I think he is asking for a log that has LAN MODE OFF

bambu123, 2023-08-09:
> > > This is the whole log directory.
> > > Thanks :)
> > > [log.zip](https://github.com/bambulab/BambuStudio/files/12297412/log.zip)
> > 
> > 
> > Sorry. I'm looking into this log. LAN only mode is on in this log.
> 
> @TheCableGuy99 I think he is asking for a log that has LAN MODE OFF so it will exhibit the issue

@TheCableGuy99 I would like to ask you to help compare whether there is a difference between the two modes.

TheCableGuy99, 2023-08-09:
Ahh okay, my bad! I turned LAN only mode back off earlier as I was having other issues with it and have done some prints since so I can provide that right away:
[log.zip](https://github.com/bambulab/BambuStudio/files/12300277/log.zip)

Also, here is a minute long video of how I am seeing the video stream and as already said it's slow to connect initially:
[](https://](https://www.privatehost.uk/video.zip)

TheCableGuy99, 2023-08-09:
Everything else on my PC and the whole network is fine. We can video stream from multiple TV/devices at once, we can watch 10 IP CCTV cameras and much more. It's all commercial grade networking switches and Wireless APs.

I've also disabled everything else on the network and it's the same showing it's not a local networking issue (but the fact that LAN mode works fine shows that also). 

PC has recently been formatted and is running very fast and the issue was present before the format and still present after and yes all other apps on the same PC are running fine. I can watch youtube and many other online TV channels just fine. Bambu Studio is also pretty quick, it's a very fast laptop.

TheCableGuy99, 2023-08-09:
> Are you saying that LAN Only mode is good? The problem only exists in non-LAN Only mode?

That's correct.

Str00pwafel, 2023-08-09:
I'm having the same issue but rather with the bambu app on phone.

Whenever I want to see progress of the print in the app, it has to load for 30 sec or more for it to show the initial picture after pressing the play button. It stays black for so long before it shows me the camera, this can't be how it supposed to be.

bambu123, 2023-08-09:
> > Are you saying that LAN Only mode is good? The problem only exists in non-LAN Only mode?
> 
> That's correct.

Can you help us debug this problem? I added some logs to BambuSource.dll. Please help replace BambuSource.dll in the plugin directory, then reproduce the problem and send us the logs. Note that this DLL is not signed, please delete it after testing.
[BambuSource.zip](https://github.com/bambulab/BambuStudio/files/12301118/BambuSource.zip)

TackleBerryCrash, 2023-08-09:
For me this started happening last couple months and got a lot worse gradually. Now sometimes it is almost unusable, sometimes barely streaming. 
This happening gradually over the months suggesting it is a capacity issue on the service end of things.
Most likely services needs to be scaled up.

TheCableGuy99, 2023-08-09:
> Can you help us debug this problem? I added some logs to BambuSource.dll. Please help replace BambuSource.dll in the plugin directory, then reproduce the problem and send us the logs. Note that this [...]

it will be a few hours as its printing now so i'll do a new print with the new dll file when its finished :)

TheCableGuy99, 2023-08-09:
okay sorry for the delay, here's the new logs attached. I left it running for about 10 minutes before taking them.
[log.zip](https://github.com/bambulab/BambuStudio/files/12304476/log.zip)

Also, I don't know if this is related but I took a backup of the original BambuSource.dll file and put it on my desktop. As soon as I opened Bambu Studio 3 files appeared on my desktop. So I'm also providing those in case they are important. They are too big to add to here to so I've uploaded them again. Here's the link:
[](https://www.privatehost.uk/Desktop_Logs.zip)

Thanks.

bambu123, 2023-08-10:
> okay sorry for the delay, here's the new logs attached. I left it running for about 10 minutes before taking them. [log.zip](https://github.com/bambulab/BambuStudio/files/12304476/log.zip)
>
> Also, I [...]

Thank you very much. But strangely, there are no relevant logs I added, but the other three files are also useful, let's analyze them first.

TheCableGuy99, 2023-08-10:
Okay, I am interested why you are looking at the logs though? Do you not feel this is a congestion or other issue with your servers?

Marcel40625, 2023-08-13:
Have the same issue, with the latest updates on BambuStudio the Video Perfomance got worse and worse.

have Ubiquiti Network Gear, have multiple IP Cams and Video Streaming from Cable Box via Network, everything runs fine, except Video Feed from X1C

@TheCableGuy99 The Video Feed does not run over Bambu´s Server as far as i know :)

TheCableGuy99, 2023-08-13:
> Have the same issue, with the latest updates on BambuStudio the Video Perfomance got worse and worse.
> 
> have Ubiquiti Network Gear, have multiple IP Cams and Video Streaming from Cable Box via Network, everything runs fine, except Video Feed from X1C
> 
> @TheCableGuy99 The Video Feed does not run over Bambu´s Server as far as i know (even in Cloud Mode) :)

The issue for me has been going on for some time. I also use Ubiquiti and all my IP cams are also fine (although they are hard wired). The video feed must go through their servers because otherwise you wouldn't be able to access it externally. I turn off uPNP as it's insecure.

bambu123, 2023-08-14:
To make the issue clearer, this issue should all be about the X1 printer model.
Studio 1.7 supports LAN Only Liveview with the latest version of X1, the previous Studio (1.6) can only support X1, non-LAN Only liveview.
In non-LAN Only scenarios, there should not be much difference between the old and new versions. If you find that the old and new versions are very different, and the new version has a regression, that is the issue that needs to be analyzed and solved.
Under normal circumstances, video streams are sent through P2P and do not require server forwarding.​
In order to analyze the problem, it is recommended to provide comparative logs of the old and new versions. In particular, the logs of the old version are normal and the new version is not normal, and the total log time span of the old and new versions does not exceed 1 hour.

TheCableGuy99, 2023-08-14:
> To make the issue clearer, this issue should all be about the X1 printer model. Studio 1.7 supports LAN Only Liveview with the latest version of X1, the previous Studio (1.6) can only support X1, [...]

Okay. slightly confused as for me it's been going on a long time and it's only recently that since I've been able to show it isn't affected when using the "LAN mode" viewer which is now supported.

I've sent you as much logs as I can, what else can I do to help (if anything)?

bambu123, 2023-08-14:
> Okay. slightly confused as for me it's been going on a long time and it's only recently that since I've been able to show it isn't affected when using the "LAN mode" viewer which is now supported. [...]

LAN Only Liveview uses a direct connection to the local area network, so it is basically unaffected by the network environment. For comparison, I need your help to provide some logs played using version 1.6.

TheCableGuy99, 2023-08-14:
Okay, just to be sure, you want the logs from the 1.6 version when using cloud mode right?

I can do this for you later today, i just want to confirm which mode you need me to get the logs in.

bambu123, 2023-08-14:
> Okay, just to be sure, you want the logs from the 1.6 version when using cloud mode right?
> 
> I can do this for you later today, i just want to confirm which mode you need me to get the logs in.

Yes.
Downgrade Studio to 1.6.2 version.
Set printer to turn off the LAN Only mode, that is to use cloud mode.
Play liveview for minutes and multiple times, close studio and send me the logs.

bambu123, 2023-08-14:
@TheCableGuy99 @TackleBerryCrash @Str00pwafel @Marcel40625 
We have adjusted some server configurations. Can you help me try again?

bambu123, 2023-08-15:
@TackleBerryCrash @Str00pwafel @Marcel40625 
What about yours?

TheCableGuy99, 2023-08-15:
I just wanted to confirm that I've done a little additional testing to ensure it's thoroughly working. I had my CCTV cameras streaming to my laptop, 3 X1C's and all streams were running smoothly. I also tried the studio app on iOS and that's also quicker to connect when on WiFi and when on 4G. Whatever you have done as far as I'm concerned, I'm really pleased with. But yeah, I'm interested what others say as well.

Str00pwafel, 2023-08-15:
> @TackleBerryCrash @Str00pwafel @Marcel40625 What about yours?

Unfortunetaly no, but it is a P1P and not an X1C.
The initial loading of the footage takes about 20 seconds if not more, after it eventually loads its fine and expected behaviour.

bambu123, 2023-08-15:
> > @TackleBerryCrash @Str00pwafel @Marcel40625 What about yours?
> 
> Unfortunetaly no, but it is a P1P and not an X1C. The initial loading of the footage takes about 20 seconds if not more, after it eventually loads its fine and expected behaviour.

P1P is discussed here #2162.
Have you upgrade the firmware of the printer to version 1.4?
Are there any behavior changes from Studio 1.6.2 to 1.7.3?

Str00pwafel, 2023-08-15:
Printer is indeed on the latest firmware and I have updated bambu studio aswell.
I mainly use the phone app(iphone) to show the progress via the camera, I haven't actually tested if bambu studio is similar or faster. I can do that soon.

TackleBerryCrash, 2023-08-15:
> @TackleBerryCrash @Str00pwafel @Marcel40625 What about yours?

This cloud outage has been happening since Sunday so I was not able to test a lot but today even w/o cloud I was able stream locally within my network fine. 
I was not able to test Handy app yet. 

I'll give a day or two until cloud is up an running and report back

Marcel40625, 2023-08-15:
Had only 1 print yesterday but there it looked better, will try the next days

Str00pwafel, 2023-08-15:
@bambu123 Just checked Bambu studio and it seems a lot faster now, the app also responds way faster in showing the camera stream!

bambu123, 2023-08-16:
Glad to see LiveView is working. Thank you all for your responses.

bambu123, 2023-08-21:
Closing this issue now that the X1C related issue has been resolved.

closed 2023-08-21 by bambu123

Witawat, 2023-09-13:
now i have problems it i use x1c last firmware 1.0.6 and use studio last version 
cam stream slow and lag

how to fix

bambu123, 2023-09-13:
> now i have problems it i use x1c last firmware 1.0.6 and use studio last version cam stream slow and lag
> 
> how to fix

Please provide app logs.
The log folder can be found from the menu "Help > Open Configuration Folder".

Witawat, 2023-09-16:
> > now i have problems it i use x1c last firmware 1.0.6 and use studio last version cam stream slow and lag
> > how to fix
> 
> Please provide app logs. The log folder can be found from the menu "Help > Open Configuration Folder".

[log.zip](https://github.com/bambulab/BambuStudio/files/12640484/log.zip)

KuraiKitsune, 2023-10-02:
If I may add to this issue.
The video quality and fps has always been terrible on the Bambu Studio app for me in Windows. I do not use LAN-only mode as I appreciate the cloud connectivity.
However, the Bambu Handy app on an android phone on the same home network gets full video quality and fps.
Below are 2 videos of the same print job happening, viewed from the 2 apps. (If they work)

Bambu Studio: https://i.gyazo.com/db5b19a1b52215133ad94298a324a3a3.mp4
Bambu Handy: https://i.gyazo.com/a2873492e3ff3c684f131d83f93b070e.mp4

bambu123, 2023-10-07:
> If I may add to this issue. The video quality and fps has always been terrible on the Bambu Studio app for me in Windows. I do not use LAN-only mode as I appreciate the cloud connectivity. However, the Bambu Handy app on an android phone on the same home network gets full video quality and fps. Why can they be so different? Why has Bambu not addressed this sooner? Below are 2 videos of the same print job happening, viewed from the 2 apps.
> 
> Bambu Studio: https://i.gyazo.com/db5b19a1b52215133ad94298a324a3a3.mp4 Bambu Handy: https://i.gyazo.com/a2873492e3ff3c684f131d83f93b070e.mp4

I don't think there is a big difference between these two videos. Did you upload the wrong file?

drewzh, 2023-10-14:
@bambu123 this issue is back again. The bambu studio feed is terribly slow/laggy. Like others, my network is not the problem.

bambu123, 2023-10-16:
> @bambu123 this issue is back again. The bambu studio feed is terribly slow/laggy. Like others, my network is not the problem.

Please send me the log.

s4mmy, 2023-10-26:
Similar issues. I can never connect when on the same network (iPhone). When I go to cellular it connects but it’s slow.

drewzh, 2023-10-26:
I've since switched to lan only mode and live view is now immediate and trouble free. I simply don't have any time to diagnose further unfortunately. I'll just have to live without the iOS app.

mhwunder, 2023-11-16:
Remote monitoring via Bambu Handy for my P1P is non existent.  Barely get one still if anything at all.  This didn't used to be the case.  Need to verify how live view is when I am on my local network.

TheCableGuy99, 2023-11-17:
Just wanted to give some feedback on this...

After this ticket developed Bambu Lab changed something at their end and suddenly it started working smoothly on my laptop and iPhone for all 3 of my printers and thus proving it wasn't a problem at my end.

More recently I had a fault with a machine, and we had to replace various parts, one of the parts we replaced was the Application board in the printer ( https://wiki.bambulab.com/en/x1/maintenance/replace-ap-board ) and when we did this it resolved the original issue but caused another where the video was laggy again and only affecting this one printer.

After much going back and forth Bambu Lab replaced the board again and this time everything worked perfectly first time.

So my point is, if you are absolutely sure your network, internet or end-device isn't at fault, it might be a hardware issue with the AP Board as I personally think their servers are working fine now, but you can put it into LAN mode anyway to check it's not a server/internet issue. 

But personally, I think this ticket should be closed now.

mhwunder, 2023-11-17:
> So my point is, if you are absolutely sure your network, internet or end-device isn't at fault, it might be a hardware issue with the AP Board as I personally think their servers are working fine now, but you can put it into LAN mode anyway to check it's not a server/internet issue.

Sorry, not sure I understand how to determine if it is a hardware fault or not.

If it works while being on my LAN both in Bambu Cloud mode and LAN mode, then it is probably a hardware issue?  Or is it the other way around?  If it doesn't work in cloud or LAN mode locally, I have a hardware issue?

TheCableGuy99, 2023-11-17:
If you're sure your network isn't at fault and you have plenty of bandwidth and speed, plus good hardware and it's not working put the printer into LAN mode to take the cloud servers out of the equation. Then while connected to your home network (i.e. on WiFi) see if it's still lagging.

If it's still lagging when on WiFi and LAN mode it might be time to open a ticket to Bambu Lab support so they can examine your logs as it could have a faulty AP board.

mhwunder, 2023-11-17:
Thank you for clarifying.  On WiFi and LAN mode it is fine.  My total bandwidth is very high with my ISP and the printer and APs haven't moved since day one.  I use Unifi devices and, of course, upgrade firmware regularly.  Not that it is a perfect test but my laptop is showing 17ms ping, 217 Mbps down and 17.7 Mbps up.  But that is obviously on 5ghz.  According to my Dashboard the P1P client seems fine.  I have it always connect to the closets AP as I have had for a long time now.

[image]

mhwunder, 2023-11-20:
I did submit a ticket. Posting this for future viewers. 

Currently, there is a known bug affecting the live view footage when using the Bambu Handy app. Rest assured, our team is actively working on resolving this issue, and it should be fixed in the upcoming app update. In the meantime, you can try the following steps to mitigate the problem:

Ensure that you are using the latest 2.2 version of Bambu Handy.

Attempt to view the live view footage with Bambu Studio closed and vice versa to check if it makes a difference.

Unbind, log out, and then log back in to rebind the printer. Power cycle your mobile device and check if this resolves the issue.

Uninstall and reinstall Bambu Handy.

Additionally, you may consider disabling the timelapse feature, as it could also help resolve the issue.

padcom, 2025-07-26:
Making this problem about WiFi is bull. It's a real bug and it's soooo easy to prove:

- start the feed (it's choppy and laggy)
- click the down arrow next to "File" menu and select "Preferences"

And just like that, magically, the feed becomes super fluid and close to real-time.

Then you close the Properties window and boom! The video is back to what it was before.

So like I said, making the issue about slow network connection is a pile of crap.