bambulab/BambuStudio #2234: Please add support for SBS plastic to Bambu Studia and printer software.

DronBroks, 2023-08-09:
Please add support for SBS plastic to Bambu Studia and printer software.

XunZhangBambu, 2023-08-10:
The material is relatively niche and currently we have no plan to add it. You can try to explore the parameters yourself. It is recommended that you use the suggested parameters on the consumables label based on the parameters of TPU 95A. We hope you can get good printing results and share the printing parameters you used with us

DronBroks, 2023-08-10:
> The material is relatively niche and currently we have no plan to add it. You can try to explore the parameters yourself. It is recommended that you use the suggested parameters on the consumables [...]

Good afternoon
I understood your answer. But you didn't understand my question. I need the ability to enter a description of the plastic in the memory of the AMC and see remotely in which cells which plastic is installed. That is, I need the opportunity to add to the description of plastic of any manufacturer and any name of plastic, and not just those that you consider necessary. Many do not even know about the existence of the plastic that you offer to choose from. There's a lot more to base plastics than what you suggest.

QingZhangBambu, 2023-11-15:
Now, we provide a way to create you won filament, you can add the filament you wanted yourself.

kotmak111, 2024-01-05:
> Пожалуйста, добавьте поддержку пластика SBS (Styrol-Butadien Sopolimer) в Bambu Studia и программное обеспечение принтера.
> 
> https://www.sopytka.ru/products/prutki-dlya-3d-printera/9007/

The sizes of the Fdplast coils are not suitable for Bambulab and are constantly stuck in the amc. Don't break Bambulab

DronBroks, 2024-01-11:
> The sizes of the Fdplast coils are not suitable for Bambulab and are constantly stuck in the amc. Don't break Bambulab

They came up with the idea of ​​rewinding plastic a long time ago and I use it.