bambulab/BambuStudio #2237: problem with login

Bjornrongved, 2023-08-09:
### Bambu Studio Version

e.g.1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu lab carbon x1

### How to reproduce

1. go to the app on pc
2. try to login 
3. than you see that the window just stay white.

### Actual results

it ' s just loading. i dont get to log in 

### Expected results

log in 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-08-09:
@Bjornrongved 

could you help to upload the log file?
the guide can be found here: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

grawsom, 2023-08-09:
Don't know if this helps, I had the same problem, no log in user folder.
But under preferences (Ctrl+P)  there was no entry in first point "language", did the drop down and choose "English".
Now I get the login form..

Bjornrongved, 2023-08-10:
Hey,

Thank you, now it’s working 😊

StoneLiBambulab, 2023-08-31:
fixed in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

bambu studio loaded an unsupported language, you can change the language to English to avoid the issue.

Thanks!
Stone Li

StoneLiBambulab, 2023-10-24:
close this issue. open a new one if there is any problem.

Thanks!

Stone Li

closed 2023-10-24 by StoneLiBambulab

Krambalkis90, 2025-05-06:
whomever will face same issue, use this link. language change did not help me.  needed to reinstall Edge

https://wiki.bambulab.com/en/bambu-studio/troubleshoot/install-issue