bambulab/BambuStudio #2238: Login issue

sinhan11, 2023-08-09:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

[image]

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Open Bambus studio. 
Click on login 
A white window appears. 

### Actual results

[image]

### Expected results

Display login Credentials.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-08-09:
@sinhan11 
could you help to upload the log file?
the guide can be found here: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

grawsom, 2023-08-09:
@sinhan11 see #2240 did help there

StoneLiBambulab, 2023-08-31:
fixed in https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

bambu studio loaded an unsupported language, you can change the language to English to avoid the issue.

Thanks!
Stone Li

closed 2024-08-08 by lanewei120