bambulab/BambuStudio #2240: can not login the screen stayes white/blank

matliv17, 2023-08-09:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Enhedsnavn	MSI
Processor	12th Gen Intel(R) Core(TM) i7-12700H   2.70 GHz
Installeret RAM	16,0 GB (15,7 GB kan bruges)
Enheds-id	23199184-5737-447D-928A-7BA8433295D3
Produkt-id	00342-20884-50966-AAOEM
Systemtype	64-bit operativsystem, x64-baseret processor
Pen og touch	Denne skærm understøtter ikke input via pen og touchinput

### Printer

P1P

### How to reproduce

Install bambu studio
run setup = choose Eu
download network thing
login = white window no option for login at all and stuck there still after 1 hour.

### Actual results

Can not login

### Expected results

Able to login and use bambu lab
[image]

### Project file & Debug log uploads

Login problems
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2023-08-09:
@matliv17 
could you help to upload the log file?
the guide can be found here: https://wiki.bambulab.com/en/x1/troubleshooting/bambu-studio-log

matliv17, 2023-08-09:
Sure i have attached them

lanewei120, 2023-08-09:
@matliv17  

I still can not find it
anything missed?

matliv17, 2023-08-09:
Don´t understand?  did you not get the log files?

 

I have attached it as a zip file is this better?

 

Or can´t you find the problem? 

 

I have attached a picture of my problem in this mail as well.

 

 

 

This is what I get and nothing happends after that.

 

Br

 

Kent Adsbol

lanewei120, 2023-08-10:
@matliv17 
I mean can not see your log

glad to see it works now

it is a issue related with language setting
we will fix it in next version

StoneLiBambulab, 2023-09-11:
please try to change the language to english. we will fixed the issue.

Thanks!
Stone Li

zerberust, 2024-01-24:
Hi, I have the same problem with a white login screen and switching the language to english doesn't help. Also I have the newest version where I would asume the problem should have been fixed for a long time. Any help?
[debug_Wed_Jan_24_13_22_08_13108.log.txt](https://github.com/bambulab/BambuStudio/files/14038321/debug_Wed_Jan_24_13_22_08_13108.log.txt)

StoneLiBambu, 2024-01-25:
hi @zerberust ,
    your problem has the same phenomenon, but not the same cause.
what's your operation system version? macos or window 7 ? window 10 ? windows 11 pro?
i will try to find the cause.

Thanks!
Stone Li（旺仔）

zerberust, 2024-01-25:
Hi Stone Li,

Thanks for getting further into the topic. I'm using win11 pro. 
Also I tried different suggestions from reddit, discussing the topic
https://www.reddit.com/r/BambuLab/comments/154db9i/bambu_studio_login_issue/

Best regards

StoneLiBambu, 2024-01-29:
@zerberust ,
out tests didn't cover win11 pro, but there are a few users who are having the same issue and i will check it out.

Thanks!
Stone Li（旺仔）

zerberust, 2024-03-12:
is there any news on the topic? I used on the pc win10 pro, not win11 pro btw. 
still can't log in

tonycstech, 2024-03-16:
The issue is Microsoft Edge dependency built into the software.
They need to remove Edge dependency from the software. Microsoft Edge is a virus and most people remove it.
Thats whats creating the problem.

zerberust, 2024-03-16:
@tonycstech thanks a lot, that solved the problem. I had the MS Edge WebView blocked. After allowing it, the login info in BambuStudio showed up.

theboyknowsclass, 2024-04-01:
how did you unblock it? @zerberust

tonycstech, 2024-04-01:
> how did you unblock it? @zerberust
you install it.

zerberust, 2024-04-05:
as mentioned, in 3rd party firewall the app "Microsoft Edge WebView2" was blocked. After changing it, the login screen in bambu studio appeared

closed 2024-08-08 by lanewei120