bambulab/BambuStudio #2242: With update 1.7.3.50 the Flow Rate (pass 1 and 2) calibration buttons cause a crash on Windows 10.

stromboli005, 2023-08-09:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Processor	Intel(R) Core(TM) i7-1065G7 CPU @ 1.30GHz   1.50 GHz
Installed RAM	32.0 GB (31.6 GB usable)
System type	64-bit operating system, x64-based processor

Crash occurs a few seconds after the calibration squares are dropped on the build plate.

### Printer

N/A

### How to reproduce

1. start new project
2. click calibration button at top of screen
3. select Flow Rate (pass 1 or 2) on the drop down
4. squares get automatically dropped on plate
5. software crashes

I also tried to create my own flow rate calibration test using generic squares, but i am unable to change flow rate on a per-object setting, it appears to be fixed in the filament settings, maybe that has something to do with it?  I've never used the flow rate calibration tool on Bambu, but i've used it extensively on OrcaSlicer and it appears to be the same tool.

### Actual results

Bambu crashes, no user input accepted beforehand.

### Expected results

Calibration squares should be populated on build plate with different flow rates for each square.

### Project file & Debug log uploads

[flow rate calibration crash logs.zip](https://github.com/bambulab/BambuStudio/files/12304324/flow.rate.calibration.crash.logs.zip)
[Empty Build Plate for flow rate calibration.zip](https://github.com/bambulab/BambuStudio/files/12304334/Empty.Build.Plate.for.flow.rate.calibration.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lanewei120, 2023-08-10:
@stromboli005 

could you have a try for this [image ](https://github.com/bambulab/BambuStudio/releases/download/v01.07.03.50/Bambu_Studio_win_public-v01.07.03.50-20230810190205.exe)?

stromboli005, 2023-08-14:
Yes, that image worked!

Thank you!

SaltWei, 2023-08-16:
Maybe similar to https://github.com/bambulab/BambuStudio/issues/2299

stromboli005, 2023-08-16:
That looks very similar to my issue and the image you sent me fixed the
problem.

zhimin-zeng-bambulab, 2023-08-17:
It has been fixed in 01.07.04.52

closed 2023-08-17 by zhimin-zeng-bambulab