bambulab/BambuStudio #2244: Add Manual Control of the Motors in the AMS Unit for Troubleshooting Purposes

Forty206, 2023-08-09:
**Is your feature request related to a problem? Please describe.**
Related to filament breakages within the AMS Unit - When there is filament tripping sensors within the AMS but is stuck between the hub motor and the feeder motor.

**Describe the solution you'd like**
It would be nice to be able to control the AMS motors similar to how we have control over the extruder motor, ex: tell them to feed forward or backwards incrementally or continuously at a slow speed

**Describe alternatives you've considered**
No alternatives at this time

**Additional context**
This idea came to me while troubleshooting a breakage. While loading slot 3, the jolt that the AMS does to check slot 4 broke the filament in that slot, and when the AMS tried to retract from slot 4, neither motor was engaged with the filament, but it was triggering the sensors in the hub and slot 4's feeder. This caused the feeder to attempt to retract endlessly, which I needed to kill the power to interrupt. Once back on, the filament was still showing loaded, but the motors were not spinning endlessly. 

At this point, what I wanted to do was to feed filament in the feeder using a manual control of the motors to push the 100mm broken segment through the ptfe tubing to a point where I could remove it from the line, such as just before the hub unit on the back of the machine. 

Instead, to clear this issue, I needed to remove all 4 spools from the AMS, detach the exit PTFE tube, unscrew the base, lift the base, unhook an internal PTFE tube, remove the filament, and then undo all of the previous steps to reassemble the machine. This is a significantly more involved procedure to clear the broken segment than the procedure I would have preferred to do, but I need additional manual controls in software to be able to perform my preferred solution

Protarios, 2023-12-08:
Yes, please add this feature. The AMS is completely useless for me, as PLA filament breaks after leaving it in the AMS for more than one week every time and this means disassemlby of the whole unit every time (and I am sure the self cutting plasic screws are not made for constant unscrewing).
It would be so easy to push out the broken filament piece, maybe with some help by pushing it with some new filament, but the only thing the AMS can do is trying to pull, what does not work on a broken filament part.

Thomas0109, 2024-11-06:
Jes please add this feature, then it is no longer necessary to dismantle the entire ams. I don't understand why this hasn't been possible for a long time.

jsondag, 2026-09-24:
This is my biggest problem with my printer. Most of the time I’d be able to clear a clog by manually jogging the main feed motor.