bambulab/BambuStudio #2246: Reload From Disk Not Working

krayvis, 2023-08-09:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Printer

X1 Carbon

### How to reproduce

Load a 3mf file.
Overwrite the 3mf file with a new version.
Attempt to use the "Reload from disk" function to load the updated geometry.

### Actual results

3D model does not update.

### Expected results

New geometry should replace old geometry.

### Project file & Debug log uploads

[Board Tester.zip](https://github.com/bambulab/BambuStudio/files/12307120/Board.Tester.zip)
No crash log.

### Checklist of files to include

- [ ] Log file
- [X] Project file

Hologos, 2023-08-09:
Duplicate #2209.

DanBao-Bambu, 2023-08-22:
Please check if the gif below meets your expectations.

[image]

closed 2024-01-18 by SaltWei