bambulab/BambuStudio #2247: Inapropriate speed and extrusion during printing with infill/outer/inner settings

Kucha1, 2023-08-09:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab github releases

### OS version

Mac OS

### Printer

Bambu X1 Carbon

### How to reproduce

1. Use default setting
2. Add infill/outer/inner setting for printing

Issue happend during print of circle with 100 mm in diameter and 1 mm of thikness

### Actual results

Printer head was moved faster then in usual case, and extrusion was instable...
This led to wave profile of layer(from second layer)
[image]
 
[image]
.
[image]
[image]

### Expected results

Hopefully solve this issue...

### Project file & Debug log uploads

[daq.stl.zip](https://github.com/bambulab/BambuStudio/files/12307305/daq.stl.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2023-08-15:
are you use 0.4 nozzle, 0.2 standard prfile?

Kucha1, 2023-08-27:
Yes

QingZhangBambu, 2023-09-06:
what the bed type are you chosen in the studio? Texture?