bambulab/BambuStudio #2248: Embedded text

JeffreyBrew, 2023-08-10:
**Is your feature request related to a problem? Please describe.**
I'm not sure if this is a bug or a feature request.  IF this is actually a bug, let me know so I can re-post this as such.

I am trying to print a box.  To avoid using support, I have placed the bottom of the box against the build plate with open side of the box facing upward (see TEXT BOX, below).  I would like to embed some text into the bottom surface of the box.  When adding text to the bottom surface of the box in Bambu Studio (BS), I cannot set the thickness to 0 (zero) so the embedded text is flush with the bottom surface of the box.  The lowest BS will let me set the thickness is 0.10mm.  This makes it nearly impossible to embed text to the bottom (the surface against the build plate) of the box.  After slicing has completed, the following two outcomes result.
1. The first layer contains only the first layer of the embedded text.  (See LAYER 1, below) The second layer contains the second layer of the embedded text and the first layer of the box.  Since the print surface has lowered 0.20mm, the first layer of the box does not adhere to the print surface resulting in spaghetti.
2.  BS displays "Warning: It seems object <object_name>.stl has floating cantilever.  Please e-orient the object of enable support generation."

**Describe the solution you'd like**
Please update BS to allow the text thickness to be set to 0.00mm so the text is flush with the surface the text is embedded in.

**Additional context**

TEST BOX
[image]

LAYER 1
[image]

zhimin-zeng-bambulab, 2023-08-10:
[image]

If you want the text to be flush with the surface, you can change the text's type to modifier, and select the filament you want.

JeffreyBrew, 2023-08-12:
@zhimin-zeng-bambulab Thanks!  That worked!  How about adding that Part Type menu to the Text Shape settings dialog box?  (see TEXT SHAPE SETTINGS, below)

Oh!  Speaking of the Text Shape settings, the Embedded field needs a little attention.  The label 'Embedded depth' overlaps the field the depth is entered into.  (see TEXT SHAPE SETTINGS, below)

TEXT SHAPE SETTINGS
[image]

zhimin-zeng-bambulab, 2023-08-15:
Adding that Part Type menu to the Text Shape settings dialog box sounds like a good idea, we will consider it, thanks!
The label 'Embedded depth' overlaps the field, this is a bug, we will fix it.

JeffreyBrew, 2023-08-15:
Thanks in advance for both.

Perhenrikbroendby, 2023-11-19:
> [image]
> 
> @JeffreyBrew If you want the text to be flush with the surface, you can change the text's type to modifier, and select the filament you want.

I have also struggled with this problem for while, but not found a solution. Changing the text to modifier, still not flush with the surface, I get this message: " A G-code  path beyond plates boundaries.", then it is not possible to print.

zhimin-zeng-bambulab, 2023-11-20:
@Perhenrikbroendby  Could you share your 3mf file?

Perhenrikbroendby, 2023-11-20:
Here is the ziped 3mf file.
Slice plate #11 then you will se the message "A G-code path beyond plates boundaries."

[Spool Rewinder Modified.zip](https://github.com/bambulab/BambuStudio/files/13417803/Spool.Rewinder.Modified.zip)

JeffreyBrew, 2023-11-20:
> I have also struggled with this problem for while, but not found a solution. Changing the text to modifier, still not flush with the surface, I get this message: " A G-code path beyond plates [...]

@Perhenrikbroendby I am unable to get text completely flush with the surface I embedded the text into.  This is because BS will not let me set the Thickness to 0 (zero).  Every time I try to set Thickness to 0 (zero), it changes to 0.10 immediately after I press the Tab key.

@zhimin-zeng-bambulab 
I think the way the Thickness setting has been implemented is wrong.  Currently, the Thickness and Embedded Depth settings are dependent on each other.  Thickness and Embedded Depth settings should be independent of each other.  In other words, whatever value Thickness is set to is the thickness of the text is regardless of the value Embedded Depth is set to.

If I set Thickness to 1.00mm, the text should be exactly 1.00mm from the top surface of the text to the bottom surface at all times until it's changed.  The Embedded Depth setting should embed the text accordingly.  For example:

- If I set Embedded Depth to 0.00mm then the bottom surface of the 1.00mm thick text should be resting the surface of the shape.  The text is unembedded.

- If I set Embedded Depth to 0.60mm then the bottom surface of the 1.00mm thick text should be embedded 0.60mm below the surface of the shape with the remaining 0.40mm above the surface of the shape.  The text is partially embedded.

- If I set Embedded Depth to 1.00mm, then the bottom surface of the 1.00mm thick text should be embedded 1.00mm below the surface of the shape leaving the top surface of the text flush with the surface of the shape.  The text is flush embedded.  This is what we need.

- If I set Embedded Depth to 3.00mm, then the bottom surface of the 1.00mm thick text should be embedded 3.00mm below the surface of the shape leaving the top surface of the text 2.00mm below the surface of the shape.  The text is completely embedded.