bambulab/BambuStudio #2249: Align Model Surface to Build Plate

zeffez, 2023-08-10:
The current "Lay on Face" alignment tool appears to provide alignment planes that are surfaces of a bounding polyhedron of the model. It would be helpful if we could select faces of the actual model to align with the build plate (similar to other slicers such as Cura). This would allow the user to align model faces with slicing planes, and then critically flat surfaces (that are not necessarily parallel to the model bottom or bounding polyhedron) could be printed as one continuous layer without layer ridges.

My current work around involves realigning the model in a 3rd party software (such as Cura), exporting a new stl file with the adjusted alignment, and then import into Bambu studio. Although this works, these additional steps to the workflow get tedious pretty quickly.

Thanks!

johnflux, 2023-10-15:
@lanewei120 Here's an example:
I have this mostly flat piece:
[image]

But it lays out with the top surface not flat.  So the result is steps in the top surface:

[image]

I would like to add support to prop it up a bit so that the top surface is flat

lanewei120, 2023-10-15:
> @lanewei120 Here's an example: I have this mostly flat piece: [image]
> 
> But it lays out with the top surface not flat. So the result is steps in the top surface:
> 
> [image]
> I would like to add support to prop it up a bit so that the top surface is flat

@johnflux 
thanks

is it possible to share this stl for us to have a check?

johnflux, 2023-10-16:
Here. Rename to just .stl
[Fore_Shield.stl.txt](https://github.com/bambulab/BambuStudio/files/12912112/Fore_Shield.stl.txt)

Lex-FX, 2024-01-02:
+1 For also being able to select a specific mesh facet/surface to orientate down. Up til now, I've mostly been working with geometric objects so being able to orientate to a projected plane is invaluable. (like having 3 specific corners touch the bed)

But when it comes to printing organics in low overhang tolerant materials (ABS, HIPS, PC etc) using the rotate function is too coarse when using the handles and entering number via XYZ is difficult due to the non-carteasian nature of organics. 

Example below is of the talus bone in your ankle. It doesn't have faces anywhere. There are critical surfaces below and on top where it articulates with other bones which need to be as smooth as possible. I've added geometric features which will interface with the tibia so it becomes a functional model on a single axis. Those surfaces need to be as smooth as possible as well. And I'm printing using HIPS as it has a similar bone like colour and sheen as well as a warm organic feel. HIPS overhangs better than ABS, but not by much. 

[image]

You can see in the above picture that there aren't any selectable faces on the head of the talus.  

Below shows the orientation that produced the best print. 

[image]

[image]

[image]

You can see how making very small adjustments will greatly affect where the overhang supports will be. Said small adjustment is really tedious with current controls. 

Below is the FlashPrint interface which is simple and easy to use. Check the surface to bed box. Select the mesh triangle. Enter to confirm. 

[image]

**I get that you don't want too many faces to appear during the selection process, so maybe a toggle option to select via plane or surface or both?** 

Try it for yourself. Check Google for talus articulating facets to know which surfaces are most important.  

[Print Basemk2_Right Lower Limb QUAD SUBD2K v10_Talus - Copy.txt](https://github.com/bambulab/BambuStudio/files/13813892/Print.Basemk2_Right.Lower.Limb.QUAD.SUBD2K.v10_Talus.-.Copy.txt)

celoranta, 2024-10-06:
+1 for this feature.

My current project is a large mask.  The project requires cutting and connectors.  Once cut, the natural 'base' of the project is the cut line, but one of the halves has a connector sticking out of it, and the face-plane of this connector is not parallel with the cut plane.  

I would like to choose the cut line as the 'face' plane, and build up around the connector using supports.

[image]

IroCorp, 2024-11-07:
Hello,
I need this same feature too !!
Currently, I rotate the object manually but it's not precise
Thank you

GeekNathan, 2025-01-02:
Yeah I'd love to be able to select a surface and set an angle for that surface, like 0 degrees relative to the build surface so the top layer(s) of the object get a good clean layer without steps in it, no matter what is going on underneath.

Haidiye00, 2025-03-28:
Hello, we have upgraded the play on face feature to the latest version. Please download the latest version
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.01.50

MatRanc, 2026-08-27:
I've opened #12048, which adds this to the **Cut** tool — press "Pick face", click a facet, and the cut plane becomes coplanar with it, oriented along its normal.

Same idea as the Triangular facet mode that Lay on Face gained from this thread, applied to cutting, since that's where an exact reference plane is often most useful: the plane lands flush with the picked face and the existing Movement field then gives you a real offset ("3 mm in from this face").

Demo GIF, implementation notes and testing details are in the PR.