bambulab/BambuStudio #2253: Overhang wall speed for inner walls not only outer waa

scottbmc, 2023-08-10:
**Is your feature request related to a problem? Please describe.**
Since the "slow down for overhangs" option is only applied to outer walls it sometimes causes curling especially with ABS.

**Describe the solution you'd like**
An option to apply the "slow down" for inner walls too

**Additional context**
This is also a feature request on Orca Slicer and a feature which is already kind of implemented in Prusa Slicer
https://github.com/SoftFever/OrcaSlicer/issues/998

QingZhangBambu, 2023-08-11:
Could you describe your issue in more detail to help me know what you want?

scottbmc, 2023-08-11:
[image]
Ok i see what you mean. I loaded an overhang test and it works only for one inner wall and only if the overhang seems to be at a specific angle.

QingZhangBambu, 2023-08-15:
in most cases, the preview layer could support the inner wall, so it would not be an overhanging wall.

QingZhangBambu, 2023-08-15:
if the inner not be fully supported, it would be detected