bambulab/BambuStudio #2258: Reload from disk not working in the latest version(s) of Bambu Studio on macOS

Erin-Mounts, 2023-08-10:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 13.4.1 (c)

### Additional system information

CPU: Apple M2
Memory: 8 GB
Display Card: Apple M2

### Printer

Bambu X1 Carbon

### How to reproduce

1. Open .3mf file containing multiple plates, each with 1 or more STEP files placed on them
2. Right-click an object and select "Reload from disk" from the dropdown menu

### Actual results

Either nothing happens (seemed to already know which file to reload from but didn't reload latest version), or after selecting the file, it pops up an alert that says "Unable to reload:" and then the selected filename.

### Expected results

Either the object already "knows" which file it came from and reloads the latest version from disk, or it doesn't and first asks you to select the file and then reloads that version from disk.

### Project file & Debug log uploads

[V118 0.2 nozzle TOS pilot laser pistol.3mf.zip](https://github.com/bambulab/BambuStudio/files/12316993/V118.0.2.nozzle.TOS.pilot.laser.pistol.3mf.zip)
[log.zip](https://github.com/bambulab/BambuStudio/files/12317006/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2023-09-15:
This error seems to be due to the step file being included in a 3mf file. After selecting a 3mf file that contains multiple step files, the program does not know which volume to load. You can try to separate individual step files for loading.

varionix, 2023-11-21:
You can use the "replace with STL" Option. This allows also to replace other file-types i.e.  *.stp, *.step, ... But you must select the new (updated) Step file with its (old/unchanged) name. 
This works for me (SW Version 1.7.6.92). Other Versions has to be tested by yourself

DanBao-Bambu, 2023-11-23:
@Erin-Mounts. Hello, the latest 1.8 Beta Bambu Studio can solve your aforementioned problem.

closed 2024-01-18 by DanBao-Bambu