bambulab/BambuStudio #2259: Developer Mode Resets to “not enabled”

Merlwynd, 2023-08-10:
### Bambu Studio Version

1.7.3

### Where is the application from?

Bambu Lab Official website

### OS version

Win10

### Printer

X1CC

### How to reproduce

Turn Developer Mode on
Close Bambu Studio
Open Bambu Studio

### Actual results

When you open Bambu Studio, Developer Mode has been turned off…even though you enabled it before quitting Bambu Studio.

### Expected results

Developer Mode should stay on when enabled, even though you close/quit Bambu Studio…similar to enabling Advanced Mode.
Not sure if this is a bug or feature request.

### Project file & Debug log uploads

Not required….

### Checklist of files to include

- [ ] Log file
- [ ] Project file

DanBao-Bambu, 2023-08-17:
I am very sorry for not incorporating the changes into this release due to my negligence. Sorry again, I will definitely pay attention next time.

tsmith35, 2023-09-12:
@DanBao-Bambu will this be corrected with the next version? It doesn't work on OrcaSlicer either, but I know you guys share source between slicers. Thanks!

DanBao-Bambu, 2023-09-12:
Thank you for your attention. In the next version, the Development Model status will be maintained after restarting Studio.

closed 2023-10-24 by bambu123