bambulab/BambuStudio #2262: [Feature Request] Improvement on plate name editing workflow

BambuLabDiscord, 2023-08-11:
1. When a user clicks at the plate name editing symbol, the cursor should be automatically placed inside the text box when the editing window opens, so user can start typing without having to use the mouse to place the cursor in first. 

2. When user finishes editing the name, hitting the Return key should save the name and exit the editing process, without requiring the user to use the mouse to click at the 'OK' button.

SaltWei, 2023-08-14:
> 1. When a user clicks at the plate name editing symbol, the cursor should be automatically placed inside the text box when the editing window opens, so user can start typing without having to use the mouse to place the cursor in first.

Good!

SaltWei, 2023-08-14:
> 2\. When user finishes editing the name, hitting the Return key should save the name and exit the editing process, without requiring the user to use the mouse to click at the 'OK' button.

We will make "enter" bottom as saving and closing

Haidiye00, 2023-08-14:
Received your request, I will handle it today

Haidiye00, 2023-08-14:
I have resolved the issue and this function will be released in version 1.8. Thank you

closed 2024-01-16 by Haidiye00