bambulab/BambuStudio #2263: Mesh Booleans still not predictable...

tubbybtch, 2023-08-11:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C w/AMS

### How to reproduce

1. Add Primitive -> Box
2. Add Primitive -> Cylinder
3. Select Both ->  Assemble
4. Mesh Boolean tool
5. Part 1 [Select] Box
6. Part 2 [Select] Cylinder
7. Difference tab
8. [Difference]

Resultant mesh is huge and distorted.

### Actual results

[image]
[image]

### Expected results

The resultant mesh should be same dimensions as Part 1 with Part 2 removed.

### Project file & Debug log uploads

[Test.zip](https://github.com/bambulab/BambuStudio/files/12318584/Test.zip)

Video included...

### Checklist of files to include

- [ ] Log file
- [X] Project file

dragon-eyes-bbl, 2023-08-11:
Thanks for the report, I can replicate the issue here.
Will update you once the issue is fixed.

dragon-eyes-bbl, 2023-08-11:
The issue will be fixed in the next release. Thanks.

SaltWei, 2023-08-17:
Included in latest release https://github.com/bambulab/BambuStudio/releases/tag/v01.07.04.52.

closed 2023-08-17 by SaltWei