bambulab/BambuStudio #2265: [Feature Request] Turn off automatic filament unload after a print and/or a prototype mode

BambuLabDiscord, 2023-08-11:
When I am working on a prototype and doing a lot of 5-10 min prints, I would like the option to turn off the filament unload after a print. I'm doing a quick print to check the dimensions and need to do another one right after. Unloading the filament wastes time and filament. I have been using the manual side loading to solve this but this is not an optimal solution. Having a prototyping mode would be great to stop unloading the filament and maybe also remove all of the pre-print stuff that takes 5-6 minutes before the print begins. I'm not talking about the bed leveling or flow tests, I have those turned off for these quick prints, I'm talking about all of the other stuff it does before a print. Just a thought. Thanks!

GuzziRaz, 2023-08-31:
A way better temporary fix than ditching the AMS is making a profile for it, here's how:

1. In Prepare or Preview tab of BambuStudio, change settings for printer by clicking the edit icon to the right of eg. “Bambu Lab (…) 0.4 nozzle”.
2. Go to tab “**Machine G-code**”, and then section “**Machine end G-code**”.
3. Find these two lines of G-code (they're located soon after the second blank line):
```
; pull back filament to AMS
M620 S255
```
Remove those two lines and then save the setting with a new name eg. “Bambu Lab (…) 0.4 nozzle - nounload”. This will be a "User preset". The "System preset" remain unchanged so whenever you actually want it to unload, pick the System Preset before printing.

Still, I too would like to have this option among the "Print Options" in the device tab, among eg. "Enable detection of build plate position". Actually I can't really think of any situation I'd actually want it to unload though?

capsel22, 2023-09-02:
> A way better temporary fix than ditching the AMS is making a profile for it, here's how:
>
>     1. In Prepare or Preview tab of BambuStudio, change settings for printer by clicking the edit icon to [...]

Absolutely amazing. Thanks for this, already used it as a workaround.
Hope Bambu can implement just a tick box when sending a print.

MiscPrinter, 2023-11-05:
+1

Adding the option with the other "Level Bed", "Enable AMS",... to not unload would be a good implementation.

chuyskywalker, 2024-01-05:
I, too, would benefit from this significantly. 

The X1C speed is amazing, and really lends itself to quick prototyping -- having to reset the AMS every print wastes a ton of time.

rsirunesys, 2026-01-30:
I would also like this feature.  I use the AMS 2 Pro.  Currently, I change the machine start and end Gcode to prevent filament from unloading and also to prevent filament purging.  I do this because during a project where I have several different filaments, I'll print all parts of the same filament... one after the other.  When I start printing, I'll choose Load from the Device tab.  Once I'm done with one type of filament I'll choose Unload from the Device tab.  This saves a ton of time and purging.  Please considering making this an option... probably on the Device tab.  Thank you.

halfblack, 2026-06-06:
Also need that option both for AMS 2 Pro and AMS Lite.