bambulab/BambuStudio #2268: Can't edit K factor

Yeello1, 2023-08-11:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 13.5

### Printer

P1S

### How to reproduce

1. Navigate to the calibration tab.
2. Perform the manual calibration procedure.
3. Input the k factor in the material editing section.
4. Click on Confirm.

### Actual results

The K factor is not saved. It stay at 0,000.
[image]

I don't have in the calibration tab the option to edit the results.
[image]

### Expected results

The k factor should be saved and appear.

### Project file & Debug log uploads

[k factor.3mf.zip](https://github.com/bambulab/BambuStudio/files/12322240/k.factor.3mf.zip)
[Archive.zip](https://github.com/bambulab/BambuStudio/files/12322244/Archive.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

rwbot, 2023-08-13:
I'm having the same problem. Same OS and BambuStudio version, on P1P

Yeello1, 2023-08-16:
When I change the language of the Slicer to English, then I can modify the K factor. But I still don't have the option to manage the results in the calibration tab. I only have to "manual calibration" button.

lukasdebaum, 2023-08-18:
I have the same problem MacOS 13.4 newest stable BambuStudio, language german.
When I change the language to english i can save the modified K factor.

LiZ-Li-BBL, 2023-09-07:
The issue has been fixed, the K factor can be normally modified in any language. Please update BambuStudio version.

closed 2023-09-07 by LiZ-Li-BBL

spudstr, 2024-04-26:
chiming in here.. latest greatest.. new x1c.. cannot update k factor.. latest mac osx.