bambulab/BambuStudio #2270: Lamp does not turn off during low power mode

emmby, 2023-08-11:
### Bambu Studio Version

1.7.3.50

### Where is the application from?

Bambu Lab Official website

### OS version

mac

### Additional system information

N/A

### Printer

P1S

### How to reproduce

1. Turn on lamp
2. Start and complete a print

### Actual results

Lamp stays on for days

### Expected results

Lamp turns off when printer enters low power mode
https://wiki.bambulab.com/en/faq#does-the-printer-support-auto-shutdown

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

GuzziRaz, 2023-09-19:
That goes for the logo lamp as well - even if the current draw is insignificant, it should turn off when in low-power mode.

I added gcode to turn it off at end of print using `M960 S5 P0 ; turn off the logo lamp.` but I haven''t seen any such code for the chamber lamp.

longwa, 2024-01-17:
The start g-code has a command to enable the logo lamp, it seems like an obvious omission that the end g-code does not then turn it off. I'm not sure there's any other way to disable the logo lamp other than in g-code, which makes it even more annoying.

Rich-b1, 2024-03-28:
On this X1C this fixes the Logo lamp staying on after a print.

https://forum.bambulab.com/t/feature-request-turn-off-hotend-and-ams-led-lights-when-in-standby/3489/23

This should be fixed in the file:
fdm_bbl_3dp_001_common.json

It's an easy fix and should be incorporated into all future studio releases.