bambulab/BambuStudio #2271: Crash when importing specific SVG in svg.cpp

vaaronna, 2023-08-11:
### Bambu Studio Version

1.7.3

### Where is the application from?

Bambu Lab github releases

### OS version

Windows & Ubuntu

### Additional system information

Crash confirmed on Windows official version and Linux built from github

### Printer

Bambu Lab X1

### How to reproduce

Create New Project
Import supplied SVG wPb.svg

### Actual results

crash/core dump from svg.cpp:195 here:
profile_line_points.back().second = profile_line_points[0].first;
because profile_line_points is empty

### Expected results

Program should not crash

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vaaronna, 2023-08-11:
[](url)
[wPb.zip](https://github.com/bambulab/BambuStudio/files/12324198/wPb.zip)

Zipped version of file

zhimin-zeng-bambulab, 2023-10-24:
The crash problem has been fixed, but there is a problem with the shape after importing. The shape error problem is due to a problem with our internal code, but it is not easy to handle. We will continue to try to solve this problem in the future.

closed 2024-10-21 by zhimin-zeng-bambulab